jbatonnet/Rinkhals #1: Request: Adding Spoolman to Kobra 3

ducati1198, 2024-12-13:
Its works
Just need a bit of cleaning 

https://github.com/utkabobr/DuckPro-Kobra3/issues/54?fbclid=IwZXh0bgNhZW0BMQABHXbOi0ZhAgGhnCcj6jP0nx8XiQRnHql3Rrzo0BLXP0KeihPrEjsGsRzQzg_aem_wd99rvVQSVoDr-16BgVXVQ

ducati1198, 2024-12-13:
Looks like spoolman fix from kiinam can fix thing

jbatonnet, 2024-12-14:
Thank you for the feedback, I need to fix my motherboard before continuing development, but I'll try to integrate those fixes!

ducati1198, 2024-12-14:
What broken on your motherboard?

ducati1198, 2024-12-14:
Salut ou je peux downloader ton .swu rooted pour le tester.
merci

jbatonnet, 2024-12-14:
It's not broken, I messed up some commands I typed manually and I now need to reflash the firmware directly on the MB.

Anyway, this repo is fully automated, I'm using Github Actions to build any commit and to automate releases. It should allow anyone to fork or contribute and create SWUs without any dev or special setup.

You can check the last one over there: https://github.com/jbatonnet/Rinkhals/actions/workflows/build-swu.yml

m3rguez0o, 2024-12-17:
Bonjour, merci pour ton super travail!
Hi thank you very much for your work!
So i successfully installed your cfw it is working but... I may do something wrong or may be I didn't understand something....
When I boot the printer ssh and mainsail, etc... aren't accessible.
So I put the usb key with the update.swu for ssh, then I run the start-rinkhals.sh and everything is working.
If I reboot the printer, I must do all the same things again.
Is it normal?
How to make it start automatically?

jbatonnet, 2024-12-17:
Hi @m3rguez0o, thank you for your message.

The **start-rinkhals.sh** script should indeed start automatically, but here are some reasons why it may not start:
- If a **.disable-rinkhals** file exists in the USB drive (**/mnt/udisk**) or in **/useremain/rinkhals**, Rinkhals will not start, I implemented those stops in order to disable RInkhals easily in case of an issue
- The **/useremain/rinkhals/.version** file should contain the name of the version you want to auto-start. This is a way to have multiple version installed and easily switch betwwen those. If the **.version** file is missing or contains an invalid version, Rinkhals won't start
- Your **/userdata/app/gk/start.sh** and **/userdata/app/gk/k3c_restart.sh** files should have been patched during installation to include the following code. If it is absent, corrupted or using wrong CRLF instead of LF for example, Rinkhals won't start.
``` sh
# Rinkhals/begin
if [ -f /useremain/rinkhals/start-rinkhals.sh ]; then
    echo Rinkhals is installed, starting...
    chmod +x /useremain/rinkhals/start-rinkhals.sh
    /useremain/rinkhals/start-rinkhals.sh
fi
# Rinkhals/end
```

At this point, you will find traces of the startup in the **/useremain/rinkhals//rinkhals.log** file. It should tell you wether Rinkhals was disabled, the version was wrong or if it didn't start.

Another weird issue I've seen is that sometimes, the printer starts multiple times. I'm not sure why exactly.
During startup, the Rinkhals **start.sh** script creates a **.disable-rinkhals** file and deletes it when everything is started properly. If for some reason the startup didn't complete or if the printer is restarting before, the .disable-rinkhals file will be there.
But again, you should see something in the logs.

Let me know if I can do anything to stabilize the startup process.

jbatonnet, 2024-12-17:
I moved this question in a separate issue: https://github.com/jbatonnet/Rinkhals/issues/4
Check my answer over there, thank you!

jbatonnet, 2024-12-17:
No don't worry, here is how to access the SWU builds, until I create the first release.
Following the picture below, you'll end up in this page: https://github.com/jbatonnet/Rinkhals/actions/runs/12379577706

From there, download the .swu file at the bottom of the page, unzip it if needed.
Then you can follow the instructions in the README: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#how-to-install-rinkhals
Or more generally the instructions in the Wiki: https://github.com/jbatonnet/Rinkhals/wiki/Firmware#how-to-install-a-firmware

[image]

ducati1198, 2024-12-17:
thank you. i never sa that download link. quick question i have Duckpro install so i need to remove any files before installing your .swu file. btw i will do a factory install of the ocf and a factory before installing your swu. i ant to make sure i install as clean as possible

jbatonnet, 2024-12-17:
If you do have a clean install, with factory install + reset, Rinkhals should work without issues, at least it's working fine on mine.

But it should work well on other firmware as well. In fact I personally used it for quite some time on top of DuckPro.
Rinkhals is built as an overlay on top of any firmware (v2.3.5.3). When it starts, it kills existing processes (including the ones from DuckPro) and runs its own copies. I made it this way so it leaves your printer "as is" and starts only when you want, and you can even keep multiple versions of Rinkhals at the same time, making it safer to install / run / remove.

So if you install it on DuckPro, it will kill DuckPro things at startup and start Rinkhals.
If you disable it (https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#1-disable-rinkhals), the firmware you had before will start instead.

ducati1198, 2024-12-18:
After installing rinkhals and running is latest version everything look working except spoolman, i can see it and control its from the mainsail gui but when putting the gcode M555 S=id it doesn’t chose the spool. What I realize is that M555 gcode is unknown command before installing the gcode micro in printer.cfg, so probably in duckpro the M555 command is not unknown

ducati1198, 2024-12-18:
After some digging, I realize that the spoolman.py was always getting rewrite on boot. I had to put permission to read only all of the file rebooted at the file was saved with the right editing 😊. Now tje only issue is i need to reboot my spoolman server to start reading the filament. My creality doesn’t don’t that spoolman server get the reading every time

m3rguez0o, 2024-12-18:
@ducati1198 Are you able to send and print a gcode using mainsail or fluidd with your printer?

ducati1198, 2024-12-18:
> @ducati1198 Are you able to send and print a gcode using mainsail or fluidd with your printer?

Yes i can send a print with orca and on my octoapp ios app

ducati1198, 2024-12-18:
Did you remove all files from duckpro ?

m3rguez0o, 2024-12-18:
I didn't install duckpro this time...
I receveid my new motherboard then done the update 2.3.5.3 then installed rinkhals.

m3rguez0o, 2024-12-18:
For your information, the code provided there [https://github.com/jbatonnet/Rinkhals/issues/4#issuecomment-2551244417](https://github.com/jbatonnet/Rinkhals/issues/4#issuecomment-2551244417) fixed this problem.

How did you get octoapp ios app to connect to the printer? I just tried and get some errors.

jbatonnet, 2025-02-08:
Release [20250208_01_test](https://github.com/jbatonnet/Rinkhals/releases/tag/20250208_01_test) should contain the patch in Moonraker, you would still need to customize the printer.cfg and moonraker.cfg, but I guess it should work now

closed 2025-02-11 by jbatonnet