jbatonnet/Rinkhals #2: how to completely remove Rinkhals

patrezp, 2024-12-15:
As the subject mentions, how do I completely remove Rinkhals from the Kobra? It persisted even after factory reset

jbatonnet, 2024-12-15:
The factory reset does not - in fact - resets the printer to factory settings. It cleans up some of the **/userdata** directory, but leave the rest untouched.

Rinkhals is built as an overlay on top of the base filesystem, so to get rid of it:

### Disable Rinkhals

**Method 1**: Create a **.disable-rinkhals** file on a USB drive or at this location: **/useremain/rinkhals/.disable-rinkhals**
This will prevent Rinkhals from starting.

**Method 2**: Factory reset might have done that already, but make sure your **/userdata/app/gk/start.sh** and **/userdata/app/gk/restart_k3c.sh** don't contain a `# Rinkhals/begin` section. If they do, remove the section between `# Rinkhals/begin` and `# Rinkhals/end`.

### Reboot

Reboot once Rinkhals is disabled to make sure it didn't start, so you'll be able to remove the files.

### Delete Rinkhals

Then you can delete the **/useremain/rinkhals** directory. That's it!

I'm working on some tools to easily debug / install / uninstall things on your printer, I'll make those step even easier!

jbatonnet, 2024-12-16:
Documentation is now part of this repo's README.md, and I added a lot of documentation in the Wiki as well. Closing this issue!

closed 2024-12-16 by jbatonnet