jbatonnet/Rinkhals #3: Disable the "Non official firmware code" screen

jbatonnet, 2024-12-16:
As mentionned by systemik, it's possible to patch the K3SysUi binary to disable the check: https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/5#discussioncomment-11577849

jbatonnet, 2024-12-16:
Note: to make it *more* persistent across versions, we may be able to path the isCustomFirmware function directly:
I am going to try on mine, and maybe we can find the code of this function and patch it dynamically.

[image]
[image]

Armstrong0005, 2025-01-01:
> Observação: para torná-lo _mais_ persistente entre as versões, podemos criar um caminho para a função isCustomFirmware diretamente: vou tentar no meu, e talvez possamos encontrar o código dessa função e corrigi-lo dinamicamente.
> 
> ![imagem](https://private-user-images.githubusercontent.com/13765215/396267352-a296100b-fdfa-4a38-a18e-2f094680e7a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3NzA4MDcsIm5iZiI6MTczNTc3MDUwNywicGF0aCI6Ii8xMzc2NTIxNS8zOTYyNjczNTItYTI5NjEwMGItZmRmYS00YTM4LWExOGUtMmYwOTQ2ODBlN2ExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAxVDIyMjgyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyYTNjZjEzMDc0MWYxNTRiZmU2NTYyODMwZGM2ZjZjMWMzODNjMjNjNmI4YjY5NzRmOWUzMDM4OWJmZmE5NDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.4QTBBPcuRHu2Pi2dP8r57xZtA_6v5juw7S7j8izaxtA) ![imagem](https://private-user-images.githubusercontent.com/13765215/396267455-4700841b-8466-4789-9e02-504bd5299110.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3NzA4MDcsIm5iZiI6MTczNTc3MDUwNywicGF0aCI6Ii8xMzc2NTIxNS8zOTYyNjc0NTUtNDcwMDg0MWItODQ2Ni00Nzg5LTllMDItNTA0YmQ1Mjk5MTEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAxVDIyMjgyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2ODQ4ODlkODNlMmU0MWNkNDEwOGVmZGYyOGEyYTg3YmVjMzM0OGFhODM4ZDMyZmIyYzJhNjlmM2E1NWUyOGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.H0_3c69YsjIy35u2oI2aTH70-VWEh2y9wYKKGMhddXM)

Onde coloco esse código?
E qual o a senha para acesso via ftp?

jbatonnet, 2025-01-22:
Fixed in latest release with the inclusion of Rinkhals touch UI [20250122_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250122_01)

closed 2025-01-22 by jbatonnet