jbatonnet/Rinkhals #4: RInkhals doesn't start automatically

jbatonnet, 2024-12-17:
[m3rguez0o](https://github.com/m3rguez0o) commented [1 hour ago](https://github.com/jbatonnet/Rinkhals/issues/1#issuecomment-2549031897)
Bonjour, merci pour ton super travail!
Hi thank you very much for your work!
So i successfully installed your cfw it is working but... I may do something wrong or may be I didn't understand something....
When I boot the printer ssh and mainsail, etc... aren't accessible.
So I put the usb key with the update.swu for ssh, then I run the start-rinkhals.sh and everything is working.
If I reboot the printer, I must do all the same things again.
Is it normal?
How to make it start automatically?

ducati1198, 2024-12-17:
after installing the update.swu i cant acces ssh, mainsail, fluid. ho can i make it work
thanks

jbatonnet, 2024-12-17:
I provide a SWU tool to access SSH on any firmware, even the stock one. They are listed here: SWU tools description: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools
They are built using GitHub actions over there: https://github.com/jbatonnet/Rinkhals/actions/workflows/build-swu-tools.yml
Last build is there: https://github.com/jbatonnet/Rinkhals/actions/runs/12342968818

You can always download the SSH tool .swu file, and as usual place it on a USB drive to get SSH access on port 2222.

Once connected, you can try to force start Rinkhals typing `/useremain/rinkhals/start-rinkhals.sh`.
If you want to go back to your previous printer firmware, you can disable Rinkhals typing `touch /useremain/rinkhals/.disable-rinkhals`
Those commands above should help you get control back over your printer, and of course you can still reflash another firmware to override Rinkhals.

If you want help running Rinkhals, please provide the startup logs:
- **/useremain/rinkhals/rinkhals.log**
- **/useremain/rinkhals/[VERSION]/rinkhals.log** with the version you downloaded

To get those files, type `cat path.log` and copy the content here.

ducati1198, 2024-12-18:
After several test. The ssh tool doesn’t update in the printer. I try with cfw and stock its give the 2 beeps and after nothing the file stay on the usb. The only wait i can access rinkhals folders is by installing cfw duckpro after installing rinkhals and reinstalling cfw duckpro after inget ssh access. The .disable-rinkhals is always present. After deleting i can boot of rinkhals. Printing work, but after configuring spoolman, i see spoolman but the macro M555 S=id doesn’t work. 

I tried several time to reinstall from factory to rinkhal with ssh tool and never works, only works on top of duckpro

I will post my logs soon

m3rguez0o, 2024-12-18:
Hello, here are the logs a little more...

```
root@Rockchip:/# cat /useremain/rinkhals/install.log
Tue Dec 17 23:10:46 CST 2024: Installing Rinkhals loader as it is missing
Tue Dec 17 23:10:47 CST 2024: Rinkhals installation complete, rebooting...
root@Rockchip:/# ls
var            useremain      tmp            sbin           root           proc           oem            media          lib64          lib            dev            bin            ac_app
usr            userdata       sys            run            rockchip_test  opt            mnt            linuxrc        lib32          etc            data           ac_lib
root@Rockchip:/# cd /useremain/rinkhals
root@Rockchip:/useremain/rinkhals# ls
install.log                               3a8dbab49e39bab6a44d94c8addefc3e351d35f5  start-rinkhals.sh                         start.sh.patch
root@Rockchip:/useremain/rinkhals# cat /useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5/rinkhals.log
Tue Dec 17 15:41:33 UTC 2024:
Tue Dec 17 15:41:33 UTC 2024: Starting Rinkhals...
Tue Dec 17 15:41:33 UTC 2024:  --------------------------------------------------
Tue Dec 17 15:41:33 UTC 2024: | Kobra firmware: 2.3.5.3
Tue Dec 17 15:41:33 UTC 2024: | Rinkhals version: 3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 15:41:33 UTC 2024: | Rinkhals root: /useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 15:41:33 UTC 2024:  --------------------------------------------------
Tue Dec 17 15:41:33 UTC 2024: > Fixing permissions...
Tue Dec 17 15:41:33 UTC 2024: > Starting SSH...
Tue Dec 17 15:41:33 UTC 2024: Killing 2691 (/tmp/swu-ssh/lib/ld-linux-armhf.so.3/tmp/swu-ssh/usr/sbin/dropbear-F-E-a-p2222-P/tmp/swu-ssh/dropbear.pid-r/tmp/swu-ssh/etc/dropbear/dropbear_rsa_host_key)
Tue Dec 17 15:41:35 UTC 2024: > Starting ADB...
Tue Dec 17 15:41:36 UTC 2024: > Stopping Klipper...
Tue Dec 17 15:41:36 UTC 2024: > Preparing chroot...
Tue Dec 17 15:41:36 UTC 2024: > Starting Moonraker...
Tue Dec 17 15:41:45 UTC 2024: > Starting nginx...
Tue Dec 17 15:41:53 UTC 2024: > Starting mjpg-streamer...
Tue Dec 17 15:41:57 UTC 2024: Killing 427 (./gkcam)
Tue Dec 17 15:42:07 UTC 2024: > Restarting Klipper...
Tue Dec 17 15:42:15 UTC 2024: > Cleaning up...
Tue Dec 17 15:42:15 UTC 2024: Killing 3611 (/useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5/lib/ld-linux-armhf.so.3./usr/sbin/dropbear-F-E-a-p2222-P/tmp/dropbear_debug.pid-r./etc/dropbear/dropbear_rsa_host_key)
Tue Dec 17 15:42:15 UTC 2024: Rinkhals started
Tue Dec 17 16:30:21 UTC 2024:
Tue Dec 17 16:30:21 UTC 2024: Starting Rinkhals...
Tue Dec 17 16:30:21 UTC 2024:  --------------------------------------------------
Tue Dec 17 16:30:21 UTC 2024: | Kobra firmware: 2.3.5.3
Tue Dec 17 16:30:21 UTC 2024: | Rinkhals version: 3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 16:30:21 UTC 2024: | Rinkhals root: /useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 16:30:21 UTC 2024:  --------------------------------------------------
Tue Dec 17 16:30:21 UTC 2024: > Fixing permissions...
Tue Dec 17 16:30:21 UTC 2024: > Starting SSH...
Tue Dec 17 16:30:21 UTC 2024: Killing 407 (/tmp/swu-ssh/lib/ld-linux-armhf.so.3/tmp/swu-ssh/usr/sbin/dropbear-F-E-a-p2222-P/tmp/swu-ssh/dropbear.pid-r/tmp/swu-ssh/etc/dropbear/dropbear_rsa_host_key)
Tue Dec 17 16:30:23 UTC 2024: > Starting ADB...
Tue Dec 17 16:30:24 UTC 2024: > Stopping Klipper...
Tue Dec 17 16:30:24 UTC 2024: > Preparing chroot...
Tue Dec 17 16:30:24 UTC 2024: > Starting Moonraker...
Tue Dec 17 16:30:34 UTC 2024: > Starting nginx...
Tue Dec 17 16:30:43 UTC 2024: > Starting mjpg-streamer...
Tue Dec 17 16:30:47 UTC 2024: Killing 482 (./gkcam)
Tue Dec 17 16:30:52 UTC 2024: > Restarting Klipper...
Tue Dec 17 16:30:59 UTC 2024: > Cleaning up...
Tue Dec 17 16:30:59 UTC 2024: Killing 1891 (/useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5/lib/ld-linux-armhf.so.3./usr/sbin/dropbear-F-E-a-p2222-P/tmp/dropbear_debug.pid-r./etc/dropbear/dropbear_rsa_host_key)
Tue Dec 17 16:30:59 UTC 2024: Rinkhals started
Tue Dec 17 16:51:52 UTC 2024:
Tue Dec 17 16:51:52 UTC 2024: Starting Rinkhals...
Tue Dec 17 16:51:52 UTC 2024:  --------------------------------------------------
Tue Dec 17 16:51:52 UTC 2024: | Kobra firmware: 2.3.5.3
Tue Dec 17 16:51:52 UTC 2024: | Rinkhals version: 3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 16:51:52 UTC 2024: | Rinkhals root: /useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5
Tue Dec 17 16:51:52 UTC 2024:  --------------------------------------------------
Tue Dec 17 16:51:52 UTC 2024: > Fixing permissions...
Tue Dec 17 16:51:52 UTC 2024: > Starting SSH...
Tue Dec 17 16:51:52 UTC 2024: Killing 927 (/tmp/swu-ssh/lib/ld-linux-armhf.so.3/tmp/swu-ssh/usr/sbin/dropbear-F-E-a-p2222-P/tmp/swu-ssh/dropbear.pid-r/tmp/swu-ssh/etc/dropbear/dropbear_rsa_host_key)
Tue Dec 17 16:51:54 UTC 2024: > Starting ADB...
Tue Dec 17 16:51:55 UTC 2024: > Stopping Klipper...
Tue Dec 17 16:51:55 UTC 2024: > Preparing chroot...
Tue Dec 17 16:51:55 UTC 2024: > Starting Moonraker...
Tue Dec 17 16:52:05 UTC 2024: > Starting nginx...
[... 575 lines cut ...]
      ██████████████████    ██

 --------------------------------------------------
| Kobra firmware: 2.3.5.3
| Rinkhals version: 3a8dbab49e39bab6a44d94c8addefc3e351d35f5
| Rinkhals root: /useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5
 --------------------------------------------------

> Fixing permissions...
> Starting SSH...
[... cut ...] id-r/tmp/swu-ssh/etc/dropbear/dropbear_rsa_host_key)
> Starting ADB...
> Stopping Klipper...
> Preparing chroot...
> Starting Moonraker...
> Starting nginx...
> Starting mjpg-streamer...
Killing 438 (./gkcam)
> Restarting Klipper...
> Cleaning up...
Killing 4043 (/useremain/rinkhals/3a8dbab49e39bab6a44d94c8addefc3e351d35f5/lib/ld-linux-armhf.so.3./usr/sbin/dropbear-F-E-a-p2222-P/tmp/dropbear_debug.pid-r./etc/dropbear/dropbear_rsa_host_key)

Rinkhals started
root@Rockchip:/useremain/rinkhals#
```

m3rguez0o, 2024-12-18:
Another issue, when I send a gcode I doesn't send on the printer.
```

09:38:57 
$ G28
10:25:03 
$ SDCARD_PRINT_FILE FILENAME="Cube_PLA_0.28_1m2s.gcode"
10:28:10 
$ RESTART
10:28:11 
project state: Disconnect
10:32:14 
$ RESTART
10:32:15 
project state: Disconnect
10:32:29 
$ G28
10:44:08 
$ SDCARD_PRINT_FILE FILENAME="Cube_PLA_0.28_1m2s.gcode"
10:45:00 
$ SDCARD_PRINT_FILE FILENAME="Cube_PLA_0.28_1m2s.gcode"
10:45:56 
$ FIRMWARE_RESTART
10:45:58 
project state: Disconnect
10:46:21 
$ RESTART
10:46:22 
project state: Disconnect

```
I tested in Mainsail and Fluidd, it's the same.

jbatonnet, 2024-12-18:
Thank you for all the detailed dumps! This makes me think I'll provide a SWU tool to easily get a log dump.

As I see, you're using version 3a8dbab and a lot of fixes in the latest version should help your situation:
- d14c432: **start.sh.patch** was using CRLF instead of LF, preventing Rinkhals to start ptoperly
- 5e74d1a: Moonraker was fixed as it was not working properly (bed mesh support, ...)
- 8c03282: Better debug output
- a0e57f3: LAN mode detection as print may not start with LAN mode disabled (now it should either way), Fix for gcodes not being found in the right path while printing

Now the annoying part, in order to the mistake I fixed in d14c432 cannot be patched using a simple install, I messed up the line endings working on Windows. To fix it, using SSH:

``` sh
cd /userdata/app/gk
cp start.sh start.sh.bak
sed 's/\r$//' start.sh.bak > start.sh
cp restart_k3c.sh restart_k3c.sh.bak
sed 's/\r$//' restart_k3c.sh.bak > restart_k3c.sh
```

This will remove the CRLF line endings and replace them with LF only.
Then you can install the latest release and all the issues listed in here should be fixed!

m3rguez0o, 2024-12-18:
> Thank you for all the detailed dumps! This makes me think I'll provide a SWU tool to easily get a log dump.
> 
> As I see, you're using version [3a8dbab](https://github.com/jbatonnet/Rinkhals/commit/3a8dbab49e39bab6a44d94c8addefc3e351d35f5) and a lot of fixes in the latest version should help your situation:
> 
>     * [d14c432](https://github.com/jbatonnet/Rinkhals/commit/d14c4321522a4147204157729a5bf88bd4395b95): **start.sh.patch** was using CRLF instead of LF, preventing Rinkhals to start ptoperly
> 
>     * [5e74d1a](https://github.com/jbatonnet/Rinkhals/commit/5e74d1a1f5231521256396e9ca884f318b75eb96): Moonraker was fixed as it was not working properly (bed mesh support, ...)
> 
>     * [8c03282](https://github.com/jbatonnet/Rinkhals/commit/8c03282f569a55ea7612ecbd8dc53108fc81a9a1): Better debug output
> 
>     * [a0e57f3](https://github.com/jbatonnet/Rinkhals/commit/a0e57f320b5b1a81570af3ea9ace61d119391669): LAN mode detection as print may not start with LAN mode disabled (now it should either way), Fix for gcodes not being found in the right path while printing
> 
> 
> Now the annoying part, in order to the mistake I fixed in [d14c432](https://github.com/jbatonnet/Rinkhals/commit/d14c4321522a4147204157729a5bf88bd4395b95) cannot be patched using a simple install, I messed up the line endings working on Windows. To fix it, using SSH:
> 
> ```shell
> cd /userdata/app/gk
> cp start.sh start.sh.bak
> sed 's/\r$//' start.sh.bak > start.sh
> cp restart_k3c.sh restart_k3c.sh.bak
> sed 's/\r$//' restart_k3c.sh.bak > restart_k3c.sh
> ```
> 
> This will remove the CRLF line endings and replace them with LF only. Then you can install the latest release and all the issues listed in here should be fixed!

Thanks! Now it works even after restarting the printer.
I can also send a gcode directly through orcaslicer.
I got an error regarding logs because the useremain/app/gk/printer_data/logs folder did not exist.
I created it with mkdir command and everything was fine after reboot.
It seems that the problem persists with the Klippy log which refers to the page http://192.168.1.45/server/files/klippy.log which is not accessible.

ducati1198, 2024-12-18:
My k3 doesn’t start no more after updating to the version  a0e57f3. At reboot the screen goes on with the line bit its never move. I try reflashing the stock firmware but i never ear the beep when it’s already the usb

m3rguez0o, 2024-12-18:
Wait the answer of jbatonnet and don't do anything stupid like I did before...
I believe something like this happened to him last week.

ducati1198, 2024-12-18:
When booting the motherboard led stays red. What those that mean?

m3rguez0o, 2024-12-18:
Nothing special, just that the motherboard is powered.

m3rguez0o, 2024-12-18:
Read the wiki [https://github.com/jbatonnet/Rinkhals/wiki/Boot](https://github.com/jbatonnet/Rinkhals/wiki/Boot) and I advise you to wait for jbatonnet’s answer. He managed to restore his bricked board.

jbatonnet, 2024-12-19:
Hi @ducati1198, the installation of Rinkhals should not be able to trigger this kind of issues in theory. The only "real" impact during installation is to append the content of [start.sh.patch](https://github.com/jbatonnet/Rinkhals/blob/master/files/start.sh.patch) to your existing **/userdata/app/gk/start.sh** and **/userdata/app/gk/restart_k3c.sh**. Appart from that I'm just copying new files around.

Now if your printer only shows the Anycubic logo with an empty progress bar any nothing else and SSH / ADB are not working, then it means that either there's something wrong physically or more probably the startup sequence is messed up somehow.

I bricked mine twice playing with commands in SSH and modifying / deleting files while exploring things. Not fun but definitely recoverable, that's why I consolidated some documentation.

After the pritner boot, it runs **/userdata/app/kenc/run.sh** that will enventually run **/userdata/app/gk/start.sh** that is responsible to run gklib, gkapi, gkcam and K3SysUi. Rinkhals startup happens at the end of all that, starting SSH, killing the existing process, starting Moonraker, ...
So if you only see this Anycubic screen and nothing else, something is wrong in the core sequence **run.sh** > **start.sh** > gklib, gkapi, gkcam and K3SysUi.

In this situation, and without SSH access, you need to reflash your partitions directly to the motherboard. I wrote a page here https://github.com/jbatonnet/Rinkhals/wiki/Boot as @m3rguez0o pointed out, it is merely a consolidated rewrite of https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/5 with additional personal experience and details.
Once connected, you should more likely focus on the **userdata** partition as detailed here: https://github.com/jbatonnet/Rinkhals/wiki/Boot#userdata-partition

jbatonnet, 2024-12-19:
Here are some working partitions backup I uploaded, it should be easier to recover: https://1drv.ms/f/s!Ag-0E4JXrqAlwbwtxJkvEgvxqJbI4Q?e=vYBveG

ducati1198, 2024-12-19:
Thanks for all the help
I been trying to flash but i get stuck at the part 

i git clode to my root after i cd xrock and make but get this error, try google but didn’t help. 
[image]

ducati1198, 2024-12-19:
Like I mentioned before I’m not any IT or programmer in life, I own a trailer manufacture and a transport company. Computer is more a big hobby of mine. I have my own home lab. It’s a pretty good set up. I know my way around a little bit in terminal. I’m pretty good with docker and VM.. I’m able to follow any tutorial that’s well explain like in your case but sometimes when I get stuck, it’s kinda hard for me to figure it out without any help.

jbatonnet, 2024-12-19:
Hi, I see that you tried to get some help over there as well: https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/5#discussioncomment-11612946

As [Bushmills](https://github.com/Bushmills) mentioned, the README gives you some hints, but your installation misses what you need to build xrock.
I feel like you're using Ubuntu or a Debian derived linux distro. You'll need to install at least **pkg-config** and **gcc**.
When you miss packages like those, you need to do:
``` sh
sudo apt update
sudo apt install X
```

In your case, I think you should install build-essentials to get what you need to build things, pkg-config that was missing, and `libusb-1.0-0-dev` as instructed in the README, so:
``` sh
sudo apt update
sudo apt install build-essential pkg-config libusb-1.0-0-dev
```

Then try to make again, it should create a xrock binary, you can `chmod +x xrock` it later to make it executable.

ducati1198, 2024-12-19:
after installing build-essential pkg-config libusb-1.0-0-dev got a other error 

$ make
[CC] crc16.c
crc16.c:45:1: fatal error: opening dependency file crc16.o.d: Permission denied
   45 | }
      | ^
compilation terminated.
make: *** [Makefile:63: crc16.o] Error 1

i try with a laptop ubuntu and a pi4 same result
thank again
[image]

jbatonnet, 2024-12-19:
Yeah this one I don't know.
Maybe try to make sure your entire directory has the right permissions. like `chmod -R 777 xrock`.
Or maybe try to run everything as root `sudo bash` and then from the root shell.

TBH I created the directory, git clone in it, make and it worked. I'm not using the same distro, PopOS on my laptop, but it should work the same.

ducati1198, 2024-12-19:
yes sir it work, now for next step lol

[image]

jbatonnet, 2024-12-20:
The main issue here should be fixed and the latest release should stabilize the startup even more, closing this one now

closed 2024-12-20 by jbatonnet

m3rguez0o, 2024-12-20:
> Here are some working partitions backup I uploaded, it should be easier to recover: https://1drv.ms/f/s!Ag-0E4JXrqAlwbwtxJkvEgvxqJbI4Q?e=vYBveG

Thanks, I think I may be able to unbrick my original motherboard with this!
At least I'll try...

m3rguez0o, 2024-12-20:
Hi, I need a little help please...
So after bricked my original mainboard, I messed up and flashed another firmware for rv1106 and now I want to restore this motherboard.
So I triyed to flash with your dump and it seems to be ok but I don't know how to do for the userdata and useremain part and the loop thing.

```
ubuntu@ubuntu:~/Desktop/kobra$ upgrade_tool 

---------------------Tool Usage ---------------------
Help:             H
Version:          V
Log:              LG
------------------Upgrade Command ------------------
ChooseDevice:		CD
ListDevice:		LD
SwitchDevice:		SD
UpgradeFirmware:	UF <Firmware> [-noreset]
UpgradeLoader:		UL <Loader> [-noreset] [FLASH|EMMC|SPINOR|SPINAND]
DownloadImage:		DI <-p|-b|-k|-s|-r|-m|-u|-t|-re image>
DownloadBoot:		DB <Loader>
EraseFlash:		EF <Loader|firmware>
PartitionList:		PL
WriteSN:		SN <serial number>
ReadSN:			RSN
ReadComLog:		RCL <File>
CreateGPT:		GPT <Input Parameter> <Output Gpt>
SwitchStorage:		SSD 
----------------Professional Command -----------------
TestDevice:		TD
ResetDevice:		RD [subcode]
ResetPipe:		RP [pipe]
ReadCapability:		RCB
ReadFlashID:		RID
ReadFlashInfo:		RFI
ReadChipInfo:		RCI
ReadSecureMode:		RSM
WriteSector:		WS  <BeginSec> <PageSizeK> <PageSpareB> <File>
ReadLBA:		RL  <BeginSec> <SectorLen> [File]
WriteLBA:		WL  <BeginSec> [SizeSec] <File>
EraseLBA:		EL  <BeginSec> <EraseCount> 
EraseBlock:		EB <CS> <BeginBlock> <BlokcLen> [--Force]
RunSystem:		RUN <uboot_addr> <trust_addr> <boot_addr> <uboot> <trust> <boot>
-------------------------------------------------------

ubuntu@ubuntu:~/Desktop/kobra$ upgrade_tool wl 0x00000000 0x00000040 env.img
Start to write env.img at 0x0...
Write LBA from file (100%)
ubuntu@ubuntu:~/Desktop/kobra$ upgrade_tool

---------------------Tool Usage ---------------------
Help:             H
Version:          V
Log:              LG
------------------Upgrade Command ------------------
ChooseDevice:		CD
ListDevice:		LD
SwitchDevice:		SD
UpgradeFirmware:	UF <Firmware> [-noreset]
UpgradeLoader:		UL <Loader> [-noreset] [FLASH|EMMC|SPINOR|SPINAND]
DownloadImage:		DI <-p|-b|-k|-s|-r|-m|-u|-t|-re image>
DownloadBoot:		DB <Loader>
EraseFlash:		EF <Loader|firmware>
PartitionList:		PL
WriteSN:		SN <serial number>
ReadSN:			RSN
ReadComLog:		RCL <File>
CreateGPT:		GPT <Input Parameter> <Output Gpt>
SwitchStorage:		SSD 
----------------Professional Command -----------------
TestDevice:		TD
ResetDevice:		RD [subcode]
[... 116 lines cut ...]
      MAJ:MIN  loop device major:minor number
       OFFSET  offset from the beginning
     PARTSCAN  partscan flag set
           RO  read-only device
    SIZELIMIT  size limit of the file in bytes
          DIO  access backing file with direct-io
      LOG-SEC  logical sector size in bytes

For more details see losetup(8).
ubuntu@ubuntu:~/Desktop/kobra$ losetup -f
/dev/loop16
ubuntu@ubuntu:~/Desktop/kobra$ sudo losetup --find --show userdata
/dev/loop0
losetup: userdata: failed to set up loop device: Is a directory
bash: /dev/loop0: Permission denied
ubuntu@ubuntu:~/Desktop/kobra$ sudo losetup --find --show userdata
/dev/loop16
losetup: userdata: failed to set up loop device: Is a directory
bash: /dev/loop16: Permission denied
ubuntu@ubuntu:~/Desktop/kobra$ $ mkdir -p _userdata && sudo mount -o rw /dev/loop16 _userdata
$: command not found
ubuntu@ubuntu:~/Desktop/kobra$ ^C
ubuntu@ubuntu:~/Desktop/kobra$ 

```

jbatonnet, 2024-12-20:
Ok, the dump I provided for **userdata** and **useremain** are in a different format than the other .img files.
.img files are a raw dump of the partitions, created using the `upgrade_tool rl` command.
**userdata** and **useremain** are copies of the files in the partitions, so you can copy those files in your own partitions.

To use them, you have to follow this part: https://github.com/jbatonnet/Rinkhals/wiki/Boot#modify-the-partitions, so basically:
- Dump your **userdata** partition: `sudo ./upgrade_tool rl 0x00080e40 0x00100000 output/userdata.img`
- `losetup` it: `sudo losetup --find --show output/userdata.img` (it should not give you any error, cause of the raw img format)
- Mount the partition: `mkdir -p _userdata && sudo mount -o rw /dev/loop0 _userdata`
- Extract my backup and copy all the files in your `_userdata`, do other modification as you wish
- Unmount it `umount _userdata`
- De-`losetup` it: `sudo losetup -d /dev/loop0`
- Reflash your partition: `sudo ./upgrade_tool wl 0x00080e40 0x00100000 output/userdata.img`

I'll see if I can directly upload the raw **userdata** partition .img file.

jbatonnet, 2024-12-20:
I uploaded updated **userdata.tgz** and **userdata.zip** in my shared directory here: https://1drv.ms/f/s!Ag-0E4JXrqAlwbwtxJkvEgvxqJbI4Q
The .tgz is still a copy of the files in the **userdata** partition.
The .zip file is a raw dump of the partition, the one I use to recover my printer every time I brick it. I just removed personal info.

m3rguez0o, 2024-12-20:
Hi, thank you for helping me.

I was able to flash the last userdata image you added, I only need useremain image now to be able to use this motherboard.

I disassembled and connected the screen to this board to see what was happening, I can now see the Anycubic logo, then the progress bar that loads, then at the end of the load the error 11407 appears. I’m not very far away thanks to you!

Before that I had tried the method you gave:

-Download userdata

-Mount userdata to copy the files inside

-Unmount userdata

-Flash userdata

But I couldn’t copy the files in _userdata(permissions denied) 

I also followed your method to do the same thing with useremain but I didn’t even manage to mount the image correctly...

I have an error from the step with « losetup... » it creates a loop the _useremain folder is present but there is something that does not work.

I will post the logs tomorrow to show you.

Another problem, since the wifi is not configured, I don’t know how to access it...
Is it possible to create an update.swu that copy the useremain folder diretly on the board?
Or does it need to be a flashed .img to work?

m3rguez0o, 2024-12-20:
I re-read some of your posts and thought about my problem.
The Anycubic screen and progress bar means that it execute code so I wonder if an update.swu that copy a modified wpa_supplicant.conf file that match my wifi credentials in the right folder, combined with your update.swu that give access to ssh and sftp would give me back the access and then let me upload the missing useremain files via sftp, am I wrong?

jbatonnet, 2024-12-21:
If you have 11407 now, I advise you follow this guide: https://github.com/jbatonnet/Rinkhals/wiki/Firmware#my-printer-shows-a-11407-error
I guess you'll have everything recovered then.
I strongly suggest you that as soon as you have a working printer, and maybe well configured, make full raw backups of all your partitions so it'll be super easy to fix later.

ducati1198, 2024-12-21:
If you printer work on 2.3.3.9. You can try to boot off 2.3.3.9 update the printer head firmware to 1.24 with duckpro backport. After the updating print head but back 2.3.5.3 files (gkapi, gklib, K3SysUi,……) in the folder, update tu stock 2.3.5.3 stock firmware.

ducati1198, 2024-12-21:
I add this error and that i fix it
[image]

ducati1198, 2024-12-21:
I add this error and this how i fix it
[image]

m3rguez0o, 2024-12-21:
Hello, I tried to boot my almost recovered motherboard in the printer (with all devices plugged) and it finally booted.
I was able to navigate in the menu, set the wifi, there was some missing things like the version of the firmware, the Anycubic qr code and some other things. I tried to restore to default in the menu, it rebooted, I re set the wifi settings, then tap on lan mode only and then all messed up...😅
The printer screen is blocked.
After reboot, it goes to the main screen with the print button but no touch screen anymore. I tried to make an update with original swu but it doesn't do anything, I tried somes usb keys but it don't start the update neither bip. The only thing I'm able to do is to connect to ssh with your tool, the only one that make bip the printer but I can't stfp.
So...I reinstalled my second motherboard in the printer. All is working like origin with your cfw sadly it's not the latest release so it don't boot automatically but before doing anything more I want to dump all the content like you told me.
I tried you dump tool, it don't seems to copy anything? 
I also tried to dump with the usb cable without success, I see fuzhou rockchip with lsusb but it doesn't want to work with the xrock maskrom command.
I don't remember the message but something like no rockchip device found 🤪
So I'am stuck...

m3rguez0o, 2024-12-22:
Hello, I have some other questions...
What if I copy all the files of userdata from my working motherboard with the sftp access then include them in the userdata.img you provided and then burn it in the bricked motherboard?
After this I may be able to connect to it and push the remaining useremain files.
Will it work? 
Will I have a clone ? With the same id's mac address et cetera?

jbatonnet, 2024-12-22:
If you manage to dump the partitions of your new MB in any way, it would definitely be the best.

Now if you cannot, yeah you can copy the files from one to the other, and then you would have a clone.
I think all the machine-specific files are:
- /userdata/app/gk/config/device.ini
- /userdata/app/gk/config/device_account.json
- /useremain/dev/device_id
- /useremain/dev/device_name
- /useremain/dev/machine_mac

You would need to copy all the userdata and useremain files from the good MB and copy them to the old one.
Then if you have a copy of the machine-specific files you can restore them, and it should work.

Make sure you are using the same firmware version on both sides as well.
And to modify the userdata.img, try to have a losetup/mount setup working to easily change files.

m3rguez0o, 2024-12-23:
Thank you for your message.