jbatonnet/Rinkhals #6: Mainsail issues?

m3rguez0o, 2024-12-18:
Hi, I think that those are mainsail specific but I post here to let you know:
First of all, if I slice and upload the file with anycubic slicer next, everything is ok and I can check what is happening on mainsail.
But when I slice with orca and then send the gcode with orca : 
 - I have problems with z offset (once it was smashing the head on the bed, the other time it was printing in the air some spaghettis)
 - The screen on the printer don't change and I can navigate in the menu but can't see the progress of the print
 - In the mainsail terminal, i see an "Unknown command: M73"
 - The printer make a pause before printing? (After waiting long time I pressed resume macro to test and the print began).

jbatonnet, 2024-12-19:
Hi, thank you for your message. For the z-offset and the pause, I'll test on my printer but that's really weird as I'm not modifying anything in this regard. I'll see if I can repro the situation.

For the screen on the printer, that's the default behavior. When you use the screen or Anycubic apps, your gcode goes **gkapi** > **gklib/Klippy**. **gkapi** is responsible to inform the printer UI **K3SysUi** to show the print screen.
Nothing to do with Moonraker here, it's a completely different path. 
When you use your slicer, your print command goes **Moonraker** > **Klippy**. 

That's why I added a Moonraker proxy that intercepts print commands, and sends the corresponding "Anycubic" command directly to **gkapi** instead.
The proxy can only work when LAN mode is enable, so make sure it is.
Again, I'll test a few things to understand if there is a missing path I need to intercept.

Finally, M73 (Show print progress) is not implemented in the printer. We might try to implement it later, but you can already see the progress on screen or in other apps. I'll just add it to the known issues list, and may just discard M73 messages later?

jbatonnet, 2024-12-20:
I added support for slicer "print" interception in moonraker-proxy. It's available in latest release https://github.com/jbatonnet/Rinkhals/releases/tag/20241219_01

I feel like the issues you mentionned about weird pritner behavior should go away with this one as well. Now the slicer cannot really trigger the print, I override it with the printer print screen directly.

Can you check the last release please? If things are working, I' ll create a separate issue or note for the M73 thingy.

ducati1198, 2024-12-22:
Confirming screen work after sending print with orca slicer.

ducati1198, 2024-12-22:
Amazing work jbatonnet. I rhink your root is better and as more features then others, let continue and support you work. Thanks. Confirming screen work after sending print with orca slicer.
Lastest version

[image]

closed 2024-12-23 by jbatonnet