jbatonnet/Rinkhals #9: Issue: need to send 2 time the print update from orca slicer

ducati1198, 2024-12-23:
Issue is that i need to send 2 time  the upload and print (orca slicer) to startup the printer. When sending the first one I see it uploading in Mainsail but doesn’t not start . Second attempt it start printing. Moonraker log 
[moonraker2.log](https://github.com/user-attachments/files/18224629/moonraker2.log)

jbatonnet, 2024-12-23:
Hi, the log you uploaded shows that you are using DuckPro right now:
data_path: /useremain/home/**ytka**/printer_data
Might be some old logs from DuckPro.

Using Rinhals, all logs are consolidated in **/useremain/rinkhals/[VERSION]/*.log**
I created a tool to easily get all logs for me or anyone to debug a situation, the SWU debug-bundle tool.

Please grab the **tools-debug-bundle.swu** tool in this page: https://github.com/jbatonnet/Rinkhals/actions/runs/12420701052
Here are the instructions on how to use: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

But basically flash it as **update.swu**, you'll hear two beeps and then you'll have a debug-bundle.zip file on the USB drive.
You can check, but there should not be any personnal or machine-specific info.
Then you can just upload the file here.

ducati1198, 2024-12-23:
Wow that weird. I got it directly on mailsail and dont have duckpro install, its a frensh install, i will upload your logs soon theres something printing right know 

My useredate folder with no duckpro:

[image]

ducati1198, 2024-12-23:
Where are your log. I got them with sftp in your folder as mentioned

[Logs.zip](https://github.com/user-attachments/files/18224779/Logs.zip)

jbatonnet, 2024-12-23:
Ok I see the 2 upload&print calls from Orca:
[image]

But I don't see any errors in the logs.
TBH, the proxy instantly sends the print command to the printer, but it can take some time before actually seeing the print start. I use it almost everyday directly from Orca, and it can take 3~6 seconds to start, but always starts on mine. Next time try to wait a little bit more.

I'll try to add more logs in the proxy, timestamps and maybe monitor the printer response to the request. There might be some clues here.

ducati1198, 2024-12-23:
Thanks
How can i change the time i. The system for montreal

jbatonnet, 2024-12-23:
I published a new release with more logs, and if the printer fails printing it will be reported and shown in Orca:
https://github.com/jbatonnet/Rinkhals/releases/tag/20241223_01

m3rguez0o, 2024-12-23:
Hello!
Just installed your latest release 20241223_01 and I have some comments.
- there was a homing /X problem on my printer it went to the right and "cracked" for some seconds and gave me an error message on the printer, I rebooted the printer, homing again in mainsail (it start automatically :) ) same thing this time I xait a little more seconds, it stop and then go to the center, it seems ok.
- when clicking print in orca, it give an internal error 500 something wrong with the server?
- if i save the gcode and then upload and print in mainsail, it re-do the wrong thing with the x axis all way right then crack with an error message on the printer and in mainsail.
- another thing I noticed, on the previous release, the camera was real time with no lag and 14/15fps, in this release it is very laggy (I have the time to go walk back to my desk, sit on my chair and I see my hand on the screen) and reach maximum 6 fps on mine.

[image]
[image]

https://github.com/user-attachments/assets/de0734c7-0955-40ee-a302-b555a500f5da

m3rguez0o, 2024-12-23:
Here it is:
[debug-bundle.zip](https://github.com/user-attachments/files/18230676/debug-bundle.zip)

I tried to go back to 20241218_01 release but the camera doesn't go over 6fps anymore...
Homing function work again.
Camera toggle in the menu don't work, it is always on.

Here is the log when starting rinkhals:

```
PowerShell 7.4.6
PS C:\Users\m3rgu> ssh root@192.168.1.45 -p2222
root@192.168.1.45's password:
root@Rockchip:~# cd ..
root@Rockchip:/# ls
var            tmp            root           oem            lib64          dev            ac_app
usr            sys            rockchip_test  mnt            lib32          data
useremain      sbin           proc           media          lib            bin
userdata       run            opt            linuxrc        etc            ac_lib
root@Rockchip:/# cd useremain
root@Rockchip:/useremain# cd rinkhals
root@Rockchip:/useremain/rinkhals# ls
20241219_01                               rinkhals.log
install.log                               20241223_01
3a8dbab49e39bab6a44d94c8addefc3e351d35f5  start.sh.patch
start-rinkhals.sh                         20241218_01
root@Rockchip:/useremain/rinkhals# ls -a
20241219_01                               start-rinkhals.sh
.current                                  rinkhals.log
.version                                  20241223_01
install.log                               .
3a8dbab49e39bab6a44d94c8addefc3e351d35f5  start.sh.patch
..                                        20241218_01
root@Rockchip:/useremain/rinkhals# cat .version
20241218_01
root@Rockchip:/useremain/rinkhals# ./start-rinkhals
-sh: ./start-rinkhals: not found
root@Rockchip:/useremain/rinkhals# ./start-rinkhals.sh

Starting Rinkhals...
Rinkhals startup was called from -sh (1381)
Rinkhals version 20241218_01 selected

Starting Rinkhals...

          ██████████
        ██          ██
        ██            ██
      ██  ██      ██  ██
      ██  ██      ██  ░░██
      ██              ░░██
        ██░░░░░░░░░░░░██
          ██████████████
      ████    ██    ░░████
    ██      ██      ░░██░░██
  ██    ██░░░░░░░░░░██  ░░░░██
  ██░░    ██████████    ░░██░░██
  ██░░                  ░░██░░██
    ██░░░░░░░░░░░░░░░░░░████░░██
      ██████████████████    ██

 --------------------------------------------------
| Kobra firmware: 2.3.5.3
| Rinkhals version: 20241218_01
| Rinkhals root: /useremain/rinkhals/20241218_01
 --------------------------------------------------

> Fixing permissions...
> Starting SSH...
Killing 1265 (/tmp/swu-ssh/lib/ld-linux-armhf.so.3/tmp/swu-ssh/usr/sbin/dropbear-F-E-a-p2222-P/tmp/swu-ssh/dropbear.pid-r/tmp/swu-ssh/etc/dropbear/dropbear_rsa_host_key)
[... 179 lines cut ...]
/dev/video10 init_mmap failed!
VIDIOC_STREAMON failed: 16, Device or resource busy
gkcam-->Nov  1 2024/14:34:02-AcCameraBusinessThread-126::Camera /dev/video10 Capture failed!
Killing 2658 (./K3SysUi)
Killing 2727 (./gkcam)
Killing 2657 (./gkapi)
Killing 2634 (./gklib-a/tmp/unix_uds1printer.cfg)
> Preparing chroot...
> Starting Moonraker...
Killing 3291 (/usr/bin/python/usr/share/moonraker/moonraker/moonraker.py)
Killing 3942 (/usr/bin/python/usr/share/scripts/moonraker-proxy.py)
> Starting nginx...
Killing 854 (nginx: master process /usr/sbin/nginx)
Killing 856 (nginx: worker process)
> Starting mjpg-streamer...
Killing 2178 (/usr/bin/mjpg_streamer-iinput_uvc.so -d /dev/video10 -n -r 1280x720-ooutput_http.so -w /usr/share/mjpg-streamer/www)
> Restarting Klipper...
> Cleaning up...
Killing 2771 (/useremain/rinkhals/20241218_01/lib/ld-linux-armhf.so.3./usr/sbin/dropbear-F-E-a-p2222-P/tmp/dropbear_debug.pid-r./etc/dropbear/dropbear_rsa_host_key)

Rinkhals started
root@Rockchip:/userdata/app/gk# cd ..
```

Do I need to delete all the files and re install it?

m3rguez0o, 2024-12-23:
these lines are red colored:
RTIsp3x          14:51:30-049 {getIspSubdevs     :100} rkisp_bypasspath entity name is NULL
RTIsp3x          14:51:30-050 {getIspSubdevs     :111} rkisp_mainpath_4x4sampling entity name is NULL
RTIsp3x          14:51:30-050 {getIspSubdevs     :122} rkisp_bypasspath_4x4sampling entity name is NULL
RTIsp3x          14:51:30-051 {getIspSubdevs     :133} isp_luma_path entity name is NULL

ducati1198, 2024-12-23:
Same issue getting the 500 error in orca. Here are logs from 20241219 ans 20241223

[debug-bundle_20241223.zip](https://github.com/user-attachments/files/18230985/debug-bundle_20241223.zip)
[debug-bundle_20241219.zip](https://github.com/user-attachments/files/18230986/debug-bundle_20241219.zip)

ducati1198, 2024-12-23:
The file get uploaded to the printer but doesn’t get started by the printer

[image]

[image]

[image]

m3rguez0o, 2024-12-23:
Here is the lag

m3rguez0o, 2024-12-23:
Same here, it needs to be added to the job que and then you can "play".
Then there is a pause you need to resume.
But at this moment there is a problem with the temperature (140° instead of what you set while slicing) so the filament is pushing on the nozzle (I saw it moving ! ) because it's not hot enough.
It's not far but there is still some issues.

m3rguez0o, 2024-12-23:
Here is the lag: 
https://we.tl/t-6rBqZrraJw

jbatonnet, 2024-12-23:
Ok, so a couple of notes:
- According to your feedbacks, I must have messed up something with moonraker-proxy. I am reinstalling the last version to better understand what's happening
- The camera lag might be caused by the proxy looping like crazy, but there might be something else. I noticed Fluidd limits to ~5fps when it's not in focus. Now even Anycubic slicer only pushes 480p, I'll try to reduce the resolution to help the printer keeping up
- The debug bundle tool collect everything but the file I needed ([VERSION]/moonraker.log), so I updated it here: https://github.com/jbatonnet/Rinkhals/actions/runs/12472190440
If one of you can send me the updated debug-bundle, that would be super helpful.

Issues with the temperature, nozzle hitting the bed or other are related to the gcode and how the printer deals with that. Rinkhals doesn't change anything about that. The only thing is RInkhals "redirects" the print function to the one using the printer screen, the rest shouldn't be impacted.

Now, the moonraker-proxy is only active on port 7125, the original moonraker is still listening on port 7126.
If you are ever stuck with the proxy not working (I mean I'll fix it but in the meantime), you can "Add another printer" to Fluidd with port 7126 to access the original moonraker instance.

[image]

jbatonnet, 2024-12-23:
Ok got it, this is what happens when I'm trying to push updates too fast.
I'll test locally and I'll release a new update with a fix for the proxy, and the 480p webcam.
With all that I hope things will work for you guys, then I'll focus on some stabilization.

m3rguez0o, 2024-12-23:
Thank you! I'll post the new logs tomorrow morning.

ducati1198, 2024-12-24:
here you go logs files ith new debug tool
[debug-bundle_3734281952_19700101-190257.zip](https://github.com/user-attachments/files/18234578/debug-bundle_3734281952_19700101-190257.zip)

ducati1198, 2024-12-24:
After updating to 20241223_02 everything looks too working fine. Print send by orca works quick, camera with no lag at 480p

ducati1198, 2024-12-24:
[image]
[image]

closed 2024-12-24 by jbatonnet

m3rguez0o, 2024-12-24:
Hello,
Here is my debug bundle
[debug-bundle_3631682348_20241224-151538.zip](https://github.com/user-attachments/files/18237016/debug-bundle_3631682348_20241224-151538.zip)

m3rguez0o, 2024-12-24:
Just updated to latest release 20241223_02, I have an issue after the reboot.
The green bar was ok then it rebooted and loaded the anycubic bar many times, then I get a 11407 error, saved the logs to the usb stick then the unofficial firmware splash, the it's stuck on the progress bar loaded at 100% but mainsail ui is accessible! 
I will now reboot the printer and I'm afraid of what will happening lol

ducati1198, 2024-12-24:
The way i update is: update to factory firmware, after use ssh tool to rm -rf rinkhals folder and then reboot and after update with root. I prefer its way because its like a fresh install

ducati1198, 2024-12-24:
> Just updated to latest release 20241223_02, I have an issue after the reboot. The green bar was ok then it rebooted and loaded the anycubic bar many times, then I get a 11407 error, saved the logs [...]

Im on my 2 print and its working only deal is the new issue I made