jbatonnet/Rinkhals #10: Printer crash and reboot after resending new print file after a print is finish

ducati1198, 2024-12-24:
After my print finish printing i resend a new print file from orca and the printer crash and reboot. After reboot i send the fille again and the print started normally using orca slicer. the log files are right after the reboot. i see this error:

[debug-bundle_3734281952_20241224-134601.zip](https://github.com/user-attachments/files/18236392/debug-bundle_3734281952_20241224-134601.zip)

1970-01-01 03:45:23,034 [shell_command.py:_check_proc_success()] - Command (/usr/bin/python /usr/share/moonraker/moonraker/components/file_manager/metadata.py -p /userdata/app/gk/printer_data/gcodes -f "gw_black_PLA_0.2_3h11m.gcode") successfully finished
1970-01-01 03:45:23,086 [application.py:log_request()] - 201 POST /api/files/local (10.11.88.200) [_TRUSTED_USER_] 9042.67ms
1970-01-01 03:45:24,440 [server.py:_process_event()] - 
Error processing callback in event job_state:state_changed
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/components/history.py", line 379, in _on_job_state_changed
    self.job_paused = job_event == JobEvent.PAUSED
                                   ^^^^^^^^^^^^^^^
AttributeError: type object 'JobEvent' has no attribute 'PAUSED'

1970-01-01 03:45:25,502 [eventloop.py:_wrapper()] - Error Running Callback
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/eventloop.py", line 89, in _wrapper
    await ret
  File "/usr/share/moonraker/moonraker/components/job_state.py", line 85, in _status_update
    self.last_event = JobEvent.from_string(new_state)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/moonraker/moonraker/common.py", line 111, in from_string
    raise ValueError(f"No enum member named {enum_name}")
ValueError: No enum member named heating
1970-01-01 03:45:26,242 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: -32601, Message: Not Found
1970-01-01 03:45:26,401 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: -32601, Message: Not Found
1970-01-01 03:45:26,953 [klippy_connection.py:_process_status_update()] - Klippy has shutdown
1970-01-01 03:45:27,189 [proc_stats.py:_handle_shutdown()] - 
Moonraker System Usage Statistics:
System Time: 13496.925759, Usage: 1.94%, Memory: 34232 kB
System Time: 13497.908960, Usage: 2.47%, Memory: 34232 kB
System Time: 13498.904627, Usage: 2.69%, Memory: 34232 kB
System Time: 13499.906490, Usage: 2.03%, Memory: 34232 kB
System Time: 13500.907491, Usage: 1.85%, Memory: 34232 kB
System Time: 13501.909950, Usage: 1.95%, Memory: 34232 kB
System Time: 13502.914001, Usage: 1.85%, Memory: 34232 kB
System Time: 13503.915833, Usage: 1.98%, Memory: 34232 kB
System Time: 13504.923646, Usage: 2.3%, Memory: 34232 kB
System Time: 13505.919582, Usage: 1.9%, Memory: 34232 kB
System Time: 13506.934975, Usage: 1.96%, Memory: 34232 kB
System Time: 13507.927879, Usage: 2.74%, Memory: 34232 kB
System Time: 13508.946474, Usage: 4.27%, Memory: 34232 kB
System Time: 13509.972611, Usage: 1.89%, Memory: 34232 kB
System Time: 13510.963907, Usage: 1.91%, Memory: 34232 kB
System Time: 13511.952569, Usage: 2.09%, Memory: 34232 kB
System Time: 13512.973126, Usage: 3.77%, Memory: 34232 kB
System Time: 13513.952347, Usage: 2.33%, Memory: 29164 kB
System Time: 13514.970108, Usage: 59.28%, Memory: 29228 kB
System Time: 13516.415765, Usage: 48.02%, Memory: 29228 kB
System Time: 13517.186310, Usage: 66.29%, Memory: 29236 kB
System Time: 13518.131319, Usage: 60.59%, Memory: 29304 kB
System Time: 13519.240628, Usage: 44.02%, Memory: 26296 kB
System Time: 13520.224116, Usage: 2.74%, Memory: 26296 kB
System Time: 13521.248195, Usage: 2.07%, Memory: 26296 kB
System Time: 13522.227892, Usage: 2.33%, Memory: 26296 kB
System Time: 13523.250741, Usage: 10.95%, Memory: 26312 kB
System Time: 13524.289915, Usage: 4.52%, Memory: 26312 kB
System Time: 13525.288163, Usage: 5.25%, Memory: 26312 kB
System Time: 13526.338905, Usage: 13.86%, Memory: 26312 kB
CPU Temperature: None
1970-01-01 03:45:27,476 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: -32601, Message: Not Found
1970-01-01 03:45:28,314 [server.py:_process_event()] - 
Error processing callback in event job_state:state_changed
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/components/history.py", line 379, in _on_job_state_changed
    self.job_paused = job_event == JobEvent.PAUSED
                                   ^^^^^^^^^^^^^^^
AttributeError: type object 'JobEvent' has no attribute 'PAUSED'

1970-01-01 03:45:31,083 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
1970-01-01 03:45:31,128 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.objects.query, Code: 503, Message: Klippy Disconnected
1970-01-01 03:45:34,463 [klippy_connection.py:_do_connect()] - Klippy Connection Established
1970-01-01 03:45:34,732 [server.py:add_warning()] - file_manager: Klipper configuration file not located in 'config' folder.

Klipper Config Path: printer.cfg

Config Folder: /userdata/app/gk/printer_data/config
1970-01-01 03:45:34,752 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
1970-01-01 03:45:34,759 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
1970-01-01 03:45:35,007 [job_state.py:_handle_started()] - Job state initialized: standby
1970-01-01 03:45:35,261 [klippy_connection.py:_verify_klippy_requirements()] - Configuration for [virtual_sdcard] not found, unable to set SD Card path
1970-01-01 03:45:35,284 [klippy_connection.py:_check_ready()] - Klippy ready
1970-01-01 03:45:35,764 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'extruder']
1970-01-01 03:45:42,459 [websockets.py:on_close()] - Websocket Closed: ID: 2726154528 Close Code: 1000, Close Reason: None, Pong Time Elapsed: 3.66
1970-01-01 03:45:42,712 [websockets.py:on_close()] - Websocket Closed: ID: 2761514992 Close Code: 1000, Close Reason: None, Pong Time Elapsed: 5.45
1970-01-01 03:45:42,738 [websockets.py:write_to_socket()] - Websocket closed while writing: 2761514992
-------------------- Log Start | Tue Dec 24 05:44:07 2024 --------------------
platform: Linux-5.10.160-armv7l-with-glibc2.39
data_path: /userdata/app/gk/printer_data
is_default_data_path: True
config_file: /userdata/app/gk/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 0475bfe679b24cab8c16140f05e093ff
unix_socket_path: /userdata/app/gk/printer_data/comms/moonraker.sock
software_version: ?
log_file: /userdata/app/gk/printer_data/logs/moonraker.log
python_version: 3.12.7 (main, Dec 11 2024, 23:16:27) [GCC 13.3.0]
launch_args: /usr/bin/python /usr/share/moonraker/moonraker/moonraker.py
msgspec_enabled: False
uvloop_enabled: False

m3rguez0o, 2024-12-24:
After 2nd print it lost connection to moonraker on mine.
When I refresh it connect but all is laggy, cam is 0fps connection to moonraker is unstable.
In the console of mainsail the unkown M73 was spamming.

[debug-bundle_3631682348_20241224-162802.zip](https://github.com/user-attachments/files/18237600/debug-bundle_3631682348_20241224-162802.zip)

ducati1198, 2024-12-24:
Print test number 3 did the same thing. I will share the logs soon for test number 3

ducati1198, 2024-12-26:
a other printer crash after sending from orca 
[debug-bundle_3734281952_20241226-110732.zip](https://github.com/user-attachments/files/18249174/debug-bundle_3734281952_20241226-110732.zip)

ducati1198, 2024-12-26:
i get this error for this print stl 

[image]

Bespinuk, 2025-01-09:
im also getting this error.

jbatonnet, 2025-01-10:
After many print attempts I was not able to repro the issue. But I noticed that in both your logs the free memory is very low.
I wonder if Moonraker + my proxy + Mainsail/Fluidd + Camera is just too much for our printers.

I've been working on some toggles to easily disable some features, so people could pick and choose what they want while leaving the rest disabled.
But the reality is that our printers' CPU and memory are quite limited.

Maybe try multiple prints from SD card, or disabling the camera in Moonraker for example to reduce memory pressure.

jbatonnet, 2025-02-08:
I didn't get any new report of this error, is it still there with the latest version ?

jbatonnet, 2025-02-11:
I believe it may have been caused by the proxy crashing with large gcodes eating memory. I'll close this one for now and will reopen if it still happens

closed 2025-02-11 by jbatonnet