jbatonnet/Rinkhals #11: Start of mjpg-streamer video stream on firmware startup

Gdenich, 2024-12-30:
Hi,
I am able to start the USB camera stream with the command:
`/useremain/dist/mjpg-streamer/mjpg_streamer -i "input_uvc.so -d /dev/video10 -n -r 1920x1080" -o "output_http.so -w /useremain/dist/mjpg-streamer/www"`

I have created the `/useremain/home/ytka/printer_data/config/.camera` file and created the script file `/useremain/home/ytka/mjpg-streamer.sh`, made i executable, and it starts the stream when executed.

But I can not start the camera upon firmware boot. Where should I put the mjpg-streamer.sh script in order to have it started?

jbatonnet, 2024-12-30:
Hi @Gdenich, you're not posting your issue in the right repo. You should open one in DuckPro firmware over there: https://github.com/utkabobr/DuckPro-Kobra3/

Rinkhals, this repo is a completely different Kobra 3 firmware. It also includes mjpg-streamer and should just work after it's flashed.

Gdenich, 2024-12-31:
Hi @jbatonnet!
In fact I am using your Rinkhals firmware from your repository, and I started to find the solutions in different sources when I found out that mjpg-streamer does not start automatically upon boot.
If it start mjpg-streamer manually (I ssh to the machine) with the command `/useremain/dist/mjpg-streamer/mjpg_streamer -i "input_uvc.so -d /dev/video10 -n -r 1920x1080" -o "output_http.so -w /useremain/dist/mjpg-streamer/www"`, as per your wiki, it starts, but I have problems with running of mjpg-streamer automatically on printer startup.

jbatonnet, 2024-12-31:
Ok my bad.
The path you mention here is from DuckPro though. I guess you installed Rinkhals on top of DuckPro then. It should work, but there might be some conflicts I'm not aware of.

Can you share the logs so I can have a look?
I made a tool to easily collect all information for debugging here: https://github.com/jbatonnet/Rinkhals/actions/runs/12472190440
And here are the instructions: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

jbatonnet, 2025-01-04:
In 20241223_02.log, I see:

```
Sun Dec 29 18:13:26 UTC 2024: > Starting mjpg-streamer...
Sun Dec 29 18:13:26 UTC 2024: Webcam /dev/video10 not found. mjpg-streamer will not start
```

But in ls-dev-v4l.log:

```
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-Jieli_Technology_USB_PHY_2.0-video-index1 -> ../../video11
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-Jieli_Technology_USB_PHY_2.0-video-index0 -> ../../video10
```

Got it thanks, I'm trying to start mjpg_streamer too fast at startup and the webcam is not ready yet.
I'm going to work on that for next release.

jbatonnet, 2025-01-08:
This issue is fixed in last release [Rinkhals 20250107_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250107_01)

closed 2025-01-08 by jbatonnet

Herbi88, 2025-01-29:
So mjep only working in fluid?