jbatonnet/Rinkhals #12: Performance issues arising from custom firmware

Yangyuanxin, 2025-01-01:
I don't think porting Klipper's original ecosystem to this platform is a good choice, as Python usually runs much slower Can you guarantee that all of this, as you have done, will not cause Timer too close issues during the printing process?

Yangyuanxin, 2025-01-02:
It's an honor to discuss this topic with you, and I'm happy and open to your suggestions for improving the firmware and selecting some good ideas to gradually optimize and import into future firmware versions. Regarding the new features you added above, I think the main goal you described is to have a similar interaction and experience as the original. I think this is a good idea, but it may need more work to support it due to performance requirements. We're very happy to build a good relationship with the community!

ducati1198, 2025-01-02:
> It's an honor to discuss this topic with you, and I'm happy and open to your suggestions for improving the firmware and selecting some good ideas to gradually optimize and import into future [...]

Hello there. The community will love for more of a open source of the factory firmware. It’s will help the community to integrate more vanilla klipper to kobra 3. I personally own a creality k1 max and the community of k1 max is up to date with all the root and mods because creality release a full open source firmware. With thatvsaid thanks for all you guys work to improve

jbatonnet, 2025-01-13:
I made a quick go app to collect and expose system metrics to Home Assistant using MQTT.
Here are the data of multiple small prints with the latest version of Rinkhals installed and enabled. I had Mainsail open on the side to create some traffic.

I intend on adding this monitoring service in the firmware in future releases to easily track performance related issues.
For now with this basic test, it seems okay, but I will leave it running long-term on my printer to see what's happening.

[image]

On top of that, I develop a small "apps" system, that allows users to easily enable or disable apps / services using the touch screen.
In next versions, I will most likely add a "memory usage" screen to show the user how much memory they have left to safely use, like in video games with VRAM usage.

I feel like it could be a solution for the long term to add features, disable them by default and allow users to pick and choose what they want based on their situation / memory availability.
CPU doesn't seem to be an issue for now, but I'm including the same principles for CPU as well for later.

[image]

Peschi90, 2025-01-26:
> It's an honor to discuss this topic with you, and I'm happy and open to your suggestions for improving the firmware and selecting some good ideas to gradually optimize and import into future [...]

But why are you not doing it then?
Still no open source as promised. A new printer before finishing the old one..... 
Man that's so annoying 
I love the printer and ams but come on....what are you guys doing. 

Rinkhals ist the first step to a real open source project but you guys throw stones in the way of the community.....

jbatonnet, 2025-01-27:
@Peschi90 I understand the frustration, but it's not useful to send this message to the devs, It's more of a business prioritization issue.

The situation is not that bad, compared to Bambu for example:
- There are already some open source components here (including go klipper): https://github.com/ANYCUBIC-3D/Kobra3
- We have the recipe to customize the firmware. I have no restrictions when working on Rinkhals for example

Now it's not perfect, but the devs have open the door to maybe collaborate on specific technical things, maybe to improve the stock firmware or to help us, I don't know yet.

If you want to help make things move, contact Anycubic's customer support about that.
It's not against any individual of course, but please do not use Rinkhals as a way to pur pressure on Anycubic in any way, this is not what I intend to achieve here with this repo.

Thank you!

jbatonnet, 2025-05-29:
See progress made in #128
I'll continue monitoring the situation closely, but it looks good :)