jbatonnet/Rinkhals #18: Issue: file_manager: Klipper configuration file not located in 'config' folder.

ducati1198, 2025-01-05:
from a fresh install i install latest version and get Klipper configuration file not located in 'config' folder.
printer still work i can print 
[image]
[image]
[debug-bundle_3906345606_20250105-103147.zip](https://github.com/user-attachments/files/18308961/debug-bundle_3906345606_20250105-103147.zip)

jbatonnet, 2025-01-05:
Hi, thank you for your message, that's a wrong warning as they are the same path.
I will fix it with next version!

ducati1198, 2025-01-08:
On lan mode in when wofi when down 2 time and to restore and reflash. Removing rinkhals folder restore the wifi. Now its looks like it working bit still have problem when i resend a print it crashed and reboot

jbatonnet, 2025-01-08:
This issue is fixed in last release [Rinkhals 20250107_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250107_01)
The other issue is being discussed in another thread

closed 2025-01-08 by jbatonnet