jbatonnet/Rinkhals #19: Wifi issue version 20250105_01

ducati1198, 2025-01-06:
after installing new version 20250105_01 i was able to do 3 print but after wifi disconnected will printing and i was never able to connect again, reboot multiple time, reflash version 20250105_01, reset wifi and nothing work. i add to reflash orignal FW and delete rinkhals folder to have wifi back. BTW on one reboot wifi came back but rinkhals didn’t start

[image]
[debug-bundle_3906345606_19700101-080106.zip](https://github.com/user-attachments/files/18312548/debug-bundle_3906345606_19700101-080106.zip)

ducati1198, 2025-01-06:
i think i found the problem its hen you turn off lan mode

jbatonnet, 2025-01-07:
Hi, I tried with and without lan mode, some prints, but I was not able to repro this issue.
Is it working now with LAN mode enabled ?

ducati1198, 2025-01-08:
yes it still happend on lan mode its just happem again the printer crash and lose ifi after reboot or it happend when only reboot by me too

ducati1198, 2025-01-08:
after a other crash in latest pre realease version, i disable rinkhald after usud ssh tool to remove every think related to rinkhals, reboot and flash to version 20241223_02 and wifi work again. il update if its crashed again

ducati1198, 2025-01-10:
After try all version after 20241223_02 they all break the wifi what ever the lan or cloud mode. They ti get back wifi is by rolling back to 20241223
[image]
[debug-bundle_3906345606_19700101-080417.zip](https://github.com/user-attachments/files/18370493/debug-bundle_3906345606_19700101-080417.zip)

jbatonnet, 2025-01-10:
I really don't get this one. I believe it but I cannot find a way to repro. I tried multiple prints, enabling and disabling lan mode, but my wifi keeps working.
Can you try to use the newer debug-bundle tool, it is able to collect logs for newer firmwares. It is available on this page: https://github.com/jbatonnet/Rinkhals/actions/runs/12642625129

The new firmwares after 20241223_02 are using binaries I rebuilt for the printer instead of the original ones, including some client binaries for the WiFi. So it must be related to this, but it is working on my printer. I must be missing something here.

I hope the new debug bundle tool will collect the logs, the interesting ones will be located in /useremain/rinkhals/20250109_01_qa/logs/*.log, including rinkhals.log and gkui.log/K3SysUi.log

Oh, maybe the specs of your WiFi are important too, can you share some details ? For example the security WPA2/3 ?

ducati1198, 2025-01-10:
after 

> I really don't get this one. I believe it but I cannot find a way to repro. I tried multiple prints, enabling and disabling lan mode, but my wifi keeps working. Can you try to use the newer [...]

i try even to connect to my phone hotshop and a travel router and add the same issue so i know its not my network
heres the logs 

[debug-bundle_3906345606_20250110-132054.zip](https://github.com/user-attachments/files/18370979/debug-bundle_3906345606_20250110-132054.zip)

ducati1198, 2025-01-10:
i dont know if its can help but i have all my access oint are unifi and my router is a netgate pfsense, i see that the prinnter get connecter to the acces point because of the signal i can see it in unifi but i cant see it on my router

jbatonnet, 2025-01-13:
I did a full factory reset of my printer (like flashing partitions directly to the MB and all), reinstalled Rinkhals and could not repro the issue.
At one point during an update I messed up I ended up with the same screen with WiFi saying connected but IP and gateway being empty. At this point I thought I could repro, but I turned WiFi off then on again on the screen and everything worked again.
In the next release I rebuilt wpa_supplicant (one of the process responsible for WiFi) with larger compatibility (EAP, WPA3, ...) in case it helps.

Gdenich, 2025-01-13:
The same problem I am struggling for a week.
I can connect to my Wireless network, for instance, three time on boot up, and then on following boots I can not connect to the wireless network six or eight boot attempts. 
At first, I thought that this is the issue with the printer wifi antenna, as if I bring the printer closer to one of the routers, I have more successful connections. I even installed a big antenna from the non-working wifi video camera. I have from -40 to -50 dB signal strength. But then it turned out that I can get the series of unsuccessful connections even near the router.
I even tried another router with 0,5 m distance, but it is the same. 

If I check the router logs (Keenetic), I can see that the printer device and router is cycling through the DHCP handshake multiple times, the router shoes the printer in the connected device list, but no connection and ping to the printer, and the printer interface shoes no IP of gateway in the interface.

I would have tried to set the static IP on the printer itself, but /etc/network/interfaces is not editable in the system.

Do not know how to troubleshoot such connection problem, as we have no access to the printer console until it is connected to the network.

Gdenich, 2025-01-13:
> I did a full factory reset of my printer (like flashing partitions directly to the MB and all), reinstalled Rinkhals and could not repro the issue.

Do you have anywhere the details of the process, maybe the factory reset could also help?

Gdenich, 2025-01-14:
> Did the problem appear with Rinkhals and specifically the last version? If so, does reinstalling the stock 2.3.5.3 or a previous Rinkhals version work?

I think the problem even persist with stock 2.3.5.3, I tried to connect to my wireless network several times, to get the IP. But I did not try use it for long, even without test print, I immediately installed 20241223_02 version, and started working with it. I think I have less problems with wireless network there, and I have more with Rinkhals 20250107_01, but maybe because I use it more often now, and I face the problem with wifi connection more frequent.

> If needed, I have some documentation on how to rewrite partitions like I just did: https://github.com/jbatonnet/Rinkhals/wiki/Boot but it's not a super easy process.

I have checked your manual, thank you! I have to analyze it more detailed to make it clear to reproduce.

jbatonnet, 2025-01-14:
I just released [20250114_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_01) with better WiFi compatibility. Pretty sure it won't fix everything since even the stock is impacted, but it can only help.

BTW the way Rinkhals works is to "overlay" its files on top of the stock firmware, so that the stock firmware remains untouched.
You can stop Rinkhals from booting having a `.disable-rinkhals` file on a USB stick for example. If the issue still appears without, a full reset might be needed maybe.

Not sure here, just trying to help with what I know..

Gdenich, 2025-01-14:
Thank you @jbatonnet! I will try the new release tomorrow. If nothing changed I will try to do the following:
- disable Rinkhals and boot the stock firmware, do a factory reset, then test the wireless connections;
If the problem persists, then:
- flash the 2.3.3.9 of the stock firmware, do a factory reset, then test the wireless connections for a good amount of boot attempts;
- flash the 2.3.5.3 version of stock, firmware, do a factory reset, then test the wireless connection;
If this does not help, then I will try the option with rewriting the partitions.

jbatonnet, 2025-01-14:
Hope it'll works, let me know!

jbatonnet, 2025-01-15:
I was investigating and I realized my overlay was hiding some stock files related to Wi-Fi. I have no idea if it can help, but I created a pre-release with some of those stock files here: [20250114_02](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_02)

ducati1198, 2025-01-15:
thanks im installing it now

Gdenich, 2025-01-15:
It's me again! Thank you @jbatonnet for the new release!

I have flashed it, but, as you expected, this did not help. 
Then I did the brainstorming, and returned my mind to the time, when during several days I had no problems with the wireless connection of the printer.

I HAVE FOUND THE PROBLEM! 

In short, this is the 40 MHz channel width for the 2.4 GHz wi-fi network. If I change the access point to work only in 20 MHz channel width, and I have 10 successful connections to the network of 10 boots! 

I have three  Keenetic routers at home, working in a mesh network. When I brought the new printer home and tried to connect the home wireless network, I had problems, and I simply switched the access point channel width to 20 MHz, and it was ok. Then one of my others devices on the network (cheap android TV box for kitchen TV) started to have slow connection speed (it is 2.4 GHz only), so I switched the channel width back to 40 MHz, tested the connection of the printer, it was working. 
So I made the conclusion, that the channel width was not a problem, it simply had temporary problems, but that was faulty.

From time to time I had connection problems on my printer, I analyzed the device behavior in my network, and found out, that the printer tried to connect not to the closest access point, but to the far access points of the mesh system. So I simply prohibited the printer to connect to other node of the mesh network, and left the connection only to the main router, that was the closest to the printer.  Starting from this moment I started to have huge problems with printer connection, but I did not have the explanation at that moment. I moved the printer this and there, changed the antenna, tried to diagnose the work of onboard rtl8723ds.

In fact, when my printer connected the farthest access points, it used the 20 MHz channel width because of distance and walls, and I had successful connection of the printer to wireless network. But when I forced the printer to connect to the closest access point, it tried to connect to the 40 MHz channel width, as the signal strength is high, but with 40 MHz channel width there is no actual data transfer, the printer makes DHCP handshake with the router, obtains the IP, but now packets go through, no ping.

This morning I thought over all these facts, and switched the access point on the main router to 20 MHz channel width, and that's it! The other access points remain 40 MHz, and I switch other devices that require 40 MHz to these nodes.

For everyone, who has problems with wireless connection of the printer, or who follows this issue, please check the channel width of your 2.4 GHz network.

jbatonnet, 2025-01-15:
Wow, great detective work, and I'm glad everything is working for you as expected now!
If you'll allow, I'll add the details in the wiki as a reference for other people.

Now if I can ask one last thing, could you possibly try with the lastest version before me adding the stock files in [20250114_02](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_02)? Those take space, and I would like to remove them if they're not useful. [20250114_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_01) would do the trick for example

Thank you for that!

Gdenich, 2025-01-15:
> Wow, great detective work, and I'm glad everything is working for you as expected now! If you'll allow, I'll add the details in the wiki as a reference for other people.
 
Yes, sure!

> Now if I can ask one last thing, could you possibly try with the lastest version before me adding the stock files in [20250114_02](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_02)? [...]

Ok I, I will revert back to 20250114_01

Gdenich, 2025-01-15:
> Now if I can ask one last thing, could you possibly try with the lastest version before me adding the stock files in [20250114_02](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_02)? [...]

I have flashed and tested the 20250114_01 again. At first I could not connect to wireless network at all, only after network reset in the printer interface, I set up new connection to the wireless network and I managed to connect again. But in this case I have only 7 successful connection of 10 attempts. And even if I have successful connection, the network interface on printer screen most of the times gives the information, that there is no connection, and even proposes to enter the wireless password again, but in fact I already have connection to the wireless network.

I flashed the 20250114_03_test release back, and it gives me 10 connections of 10 attempts, so it is really better. (I have my printer connected to the smart switch, so I switch the printer on/off in Home Assistant and check the connection in the terminal with constant ping command)

jbatonnet, 2025-01-15:
Ok thank you very much for this test. I'm not sure I fully understand why, but those files are really helping.
I'll leave them here then!

jbatonnet, 2025-01-15:
Oh yes! Perfect then, I'm closing this issue for now and hopefully we won't get any other WiFi issues
Thank you!

ducati1198, 2025-01-15:
after more testing on version 20250114_03_test wifi work good, just need to reset wifi and reboot. i did a test after a reflash and wifi not working so i connected to my guest wifi and same thing no ip but after reboot wifi work, so i log back to my main wifi and wifi not working but gateway 0.0.0.0 and ip was the one from my guest network after a reboot wifi work again and add the ip of my main wifi. mo sure if this can help

kubax, 2025-01-16:
Sadly i now have the same problem.

Neither the 03_test nor the latest 14_02 firmware work for me. Also with the latest 14_02 the custom firmware screen seems to be back?

jbatonnet, 2025-01-16:
The custom firmware screen removal comes with the new touch UI. As soon as you revert to a version like 14_02 before the UI it will be back.

For the WiFi, it is very difficult for me because no matter what I do I cannot find a way to repo this issue.

This firmware overlays its file system on top of the existing one, replacing all binaries and libraries by my own. I definitively think that's the way to go because it protects the stock firmware, allowing easy start / stop / update, ...
I use simple bind mounts to do that: https://github.com/jbatonnet/Rinkhals/blob/971d8a552a0e31a02281d8585401128b399fa7b7/files/4-rinkhals/start.sh#L116

Now there is another way, by using overlayfs but it is not available on the Kobra for now. I can try to build it but it may be a lot of effort for maybe nothing. It would allow overlaying my files, but merging them with the existing ones instead of hiding them.
It could be better, or worst. I'll read to see if there are easier options to achieve that.

And the few logs I'm dumping in the bundles are not enough to debug that. You could try to check the results of those commands, but with a spotty WiFi it is not easy. Maybe someone could try with a USB to Ethernet adapter.
- General system log: `dmesg`
- List of installed kernel modules: `lsmod`
- Some WiFi logs: `tail -f /useremain/rinkhals/.current/logs/K3SysUi.log`
- Network debugging:
    - `route print` for Routing
    - `ifconfig` or `ip a` for IP
    - `nslookup X` for DNS
    - `ping google.ca` vs `ping 8.8.8.8` for DNS
    - Play with `iw` and `wpa_supplicant` to have more information on WiFi

kubax, 2025-01-16:
I did try with a USB to Ethernet adapter, sadly without luck. Looks like not even a DHCPREQEST is send by the printer for the usb adapter.

Gdenich, 2025-01-16:
> Maybe someone could try with a USB to Ethernet adapter

I have tried the USB-Ethernet adapter as well, but I have the following for `ip a`

```root@Rockchip:/root# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq qlen 1000
    link/ether 76:56:a9:51:2e:24 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.126/24 scope global eth0
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether a8:96:09:fc:f4:44 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.125/24 brd 192.168.1.255 scope global wlan0
       valid_lft forever preferred_lft forever
4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether aa:96:09:fc:f4:44 brd ff:ff:ff:ff:ff:ff
```
It shoes `NO-CARRIER`

And `lsmod | grep usb` gives nothing.

I guess we do not have required modules for this device to work properly.

Gdenich, 2025-01-16:
Maybe there is an option to delay the startup of the wireless interface? May it be the case, that something goes wrong wireless network initialization during boot, when other parts of the firmware start.

Gdenich, 2025-01-16:
> but it seems like the usb ethernet adapter is recognized. i might try to use dhclient or something to get an ip adress via a update script. assuming dhclient is installed.

Hi @kubax ! Could you please specify, what Ethernet adapter do you use?

kubax, 2025-01-16:
@Gdenich it's a UGREEN 2.5gbit one, but it looks like the eth0 is thought was the usb adapter is a not populated device on the board itself... at least i could not get a ip address with udhcp and the mac doesn#t match my adapter.

kubax, 2025-01-16:
in my case it looks like it was PEBKAC.

As i mentioned in my first comment to this issue, i did play around with the printer.cfg. Looks like on my side gklib did throw an error, and thus network was dimissed and it didn't try to recover.

i removed the timelapse gcode from the printer.cfg and everything is working again.

the problem why i couldn't get the ssh update to work, was because i renamed the zip containering the update.swu package. must have been late.. My wifi seems to work fine, but i didn't yet try to print something again (to late here, need sleep ^^)

Yangyuanxin, 2025-01-19:
Is it possible that it is an IP renewal issue? Trying udhcpc-i wlan0 may solve this problem.

jbatonnet, 2025-01-19:
Thank you all for the debugging efforts.
I managed to repro the issue once but I don't know how exactly. I then turn off/on the WiFi through the UI, reboot the printer and it worked.

Now, I tried hard but I did not find an easy way to have overlayfs (or unionfs, mergerfs, ...) working on the printer. I had the idea of rebuilding the overlay by hand. I create a temporary directory where I symlink the base and overlay files, and then I mount this temp directory on the original one.

It allows to use all the original files except the ones I provide. In theory the printer would then use all the default WiFi binaries, libraries and configuration.
It seems to be working, so I'll create a test release soon with this new overlay system and you'll be able to test if it helps.

jbatonnet, 2025-01-20:
Alright, you should be able to try this one: [20250119_01_test](https://github.com/jbatonnet/Rinkhals/releases/tag/20250119_01_test)
I really hope this will work this time!

jbatonnet, 2025-01-21:
@kubax, @Gdenich, @ducati1198 

Would it be possible for you folks to use the latest version [20250119_01_test](https://github.com/jbatonnet/Rinkhals/releases/tag/20250119_01_test)?

If it works, I'll push to master and we'll finally have something stable.
Thank you in advance.

ducati1198, 2025-01-21:
I have it install from day one and i add no problem up to know, wifi connect fine. The only thing I don’t have the rinkhals ui. When intall version 19_01 i did a fresh factory firmware and erased everything related to rinkhals (rinkhals folder and home), did a factory restore, after install version 19_01 and everything work out of the box for me. Thanks.

kubax, 2025-01-21:
Right now i don't have any Wifi problems, and it looks like mine did come from a crashing gklib, but i will give it a try tomorrow and see if there are any problems. (I'm printing parts at the moment, so i can't upgrade right away..)

jbatonnet, 2025-01-21:
@ducati1198 I just published an updated touch UI test release based on the new overlay you're using: [20250121_01_test](https://github.com/jbatonnet/Rinkhals/releases/tag/20250121_01_test)

One the overlay is stable, I'll release the touch UI for everyone.

jbatonnet, 2025-01-22:
Ok, I released the overlay and the UI in latest release: [20250122_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250122_01)

I'm closing this issue for now as it seems to be working, thank you for your help

closed 2025-01-22 by jbatonnet