jbatonnet/Rinkhals #21: Port vor klipper

feltypaede01, 2025-01-07:
Good evening, I have just uploaded this cfg firmware but I just can't get a port over 222, 4408 etc. doesn't work either?

jbatonnet, 2025-01-07:
Hi, thank you for your message. By flashing this custom firmware on a clean printer, you should automatically get access to SSH on port 22, Mainsail on 80, and Fluidd on 4408.
Make sure you follow the installation steps on the home page and things should work out of the box 
If you ever have some trouble getting things running, please use the debug bundle SWU tool listed on the home page, it will create a zip file containing the logs and information needed to debug

feltypaede01, 2025-01-07:
Thank you very much, yes, I inserted the update swu into the folder, everything worked wonderfully and grpn lit up and the installation was complete. I have my IP from the printer but unfortunately the 22 and 80 don't work either

jbatonnet, 2025-02-11:
Hi, is this issue resolved?

MorfarJE, 2025-02-13:
I did the installation on my kobra. but it seems like it is not starting or something. it installed as descriped. with green progress bar and it rebooted fin but i cant access or see on the printer that it is installed. what am I doing wrong

jbatonnet, 2025-02-22:
Can you use the debug bundle SWU tool to get the logs from your printer? It might contain information to help diagnose the issue.
Instructions are there: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

catroc, 2025-02-23:
Same problem here ...
Logfile said all installed but nothing works.
No ssh, no Mainsail/Fluidd, no extra menu.

Swu tools don't worm too

jbatonnet, 2025-02-23:
Ok, it might happen is Rinkhals was installed but disabled. It will automatically disable itself if there was an issue during any startup.
Basically if the logs said it was installed, there must be a file called `/useremain/rinkhals/.disable-rinkhals`. This file should be removed and Rinkhals will start again.
Now, the fact that SWU tools are not working is really weird. Those tools are not connected to RInkhals in any way, the SSH tool should always work for example. If you place the update.swu on the directory on your USB key and plug it, you should hear 2 beeps. A long one and a short one.
When you hear those, it means SSH is now listening on port 2222. Connect to it using a SSH client, (user: root, password: rockchip).
Then remove the `/useremain/rinkhals/.disable-rinkhals` file and reboot.

Next version of Rinkhals will include a backup ADB server and a way to create a `.enable-rinkhals` on a USB drive to restore RInkhals startup.

catroc, 2025-02-24:
I flashed the new version today ... green bar, 2 beeps, a good logfile but nothing after reboot.

If I take a .enable-rinkhals file on my USB stick, the printer boots normal.
But ... touch doesn't work anymore and after about 30 seconds the printhead gets dark and the printer reboots.

jbatonnet, 2025-02-25:
What you describe is close to the normal boot process of RInkhals. You will see 2 boots within ~30s but then Rinkhals should be started properly. If you hear 2 close beeps during startup it means it didn't start properly.
And unless you modified the features, SSH and Mainsail will be up and running.

But without logs I cannot do anything else here, if it's not working, can you provide a debug bundle as described before?

Nigh7man, 2025-03-01:
Same problem

[debug-bundle_2907832361_20250301-163542.zip](https://github.com/user-attachments/files/19037952/debug-bundle_2907832361_20250301-163542.zip)

jbatonnet, 2025-03-01:
Okay so the issue is Moonraker configuration file is empty for some reason. If you delete it then it will be recreated properly.

ADB is running on your printer, so you can always connect using:
```
adb connect PRINTER_IP
adb shell
```

Else, you can have the .enable-rinkhals file and SSH will be running on port 22 (user: root, password; rockchip).

Then either using ADB or SSH, run:
```
rm /useremain/home/rinkhals/printer_data/config/moonraker.conf
sync
reboot
```

Then with a .enable-rinkhals file next boot you should be good to go

Nigh7man, 2025-03-01:
i did . have another problem. long boot and 11407 error

jbatonnet, 2025-03-01:
Something weird is going on with the USB drive, this is the only weird thing I see on the logs.
You might have a defective USB device or port maybe?
Can you still boot with Rinkhals disabled?

In K3SysUi.log (just before a crash or something?):
```
umount: can't unmount /mnt/udisk: Invalid argument
```

In dmesg:
```
[  135.065151] usb 1-1.3.3: USB disconnect, device number 34
[  135.303741] usb 1-1.3.3: new full-speed USB device number 35 using xhci-hcd
[  135.394442] [ac]usb_parse_endpoint:272,size=0,----
[  135.416725] cdc_acm 1-1.3.3:1.0: ttyACM0: USB ACM device
[  138.649150] usb 1-1.3.3: USB disconnect, device number 35
[  138.887085] usb 1-1.3.3: new full-speed USB device number 36 using xhci-hcd
[  138.977820] [ac]usb_parse_endpoint:272,size=0,----

... 20 to 30 times more ...

[  293.091068] [ac]usb_parse_endpoint:272,size=0,----
[  293.144709] cdc_acm 1-1.3.3:1.0: ttyACM0: USB ACM device
[  296.345152] usb 1-1.3.3: USB disconnect, device number 79
[  296.583700] usb 1-1.3.3: new full-speed USB device number 80 using xhci-hcd
[  296.674444] [ac]usb_parse_endpoint:272,size=0,----
[  296.728752] cdc_acm 1-1.3.3:1.0: ttyACM0: USB ACM device
[  299.929146] usb 1-1.3.3: USB disconnect, device number 80
[  300.167032] usb 1-1.3.3: new full-speed USB device number 81 using xhci-hcd
[  300.257825] [ac]usb_parse_endpoint:272,size=0,----
[  300.312724] cdc_acm 1-1.3.3:1.0: ttyACM0: USB ACM device
[  302.437036] usb 1-1.2: new high-speed USB device number 82 using xhci-hcd
[  302.527460] [ac]usb_parse_endpoint:272,size=0,----
[  302.530867] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[  302.533349] scsi host0: usb-storage 1-1.2:1.0
[  303.513175] usb 1-1.3.3: USB disconnect, device number 81
[  303.547780] scsi 0:0:0:0: Direct-Access     Netac    OnlyDisk         2.00 PQ: 0 ANSI: 4
[  303.548856] sd 0:0:0:0: [sda] 30720000 512-byte logical blocks: (15.7 GB/14.6 GiB)
[  303.549044] sd 0:0:0:0: [sda] Write Protect is off
[  303.549060] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[  303.549229] sd 0:0:0:0: [sda] No Caching mode page found
[  303.549244] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  303.552459]  sda: sda1
[  303.554262] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

kwe1009, 2025-03-08:
I am having the same issue.  It's as if the installation failed but I don't think it did.  Thanks for your help.

[debug-bundle_3907738554_20250309-023631.zip](https://github.com/user-attachments/files/19145319/debug-bundle_3907738554_20250309-023631.zip)

jbatonnet, 2025-03-08:
@kwe1009 Rinkhals is not installed properly, a lot of files are missing. You might have interrupted the installation somehow.

kwe1009, 2025-03-08:
> [@kwe1009](https://github.com/kwe1009) Rinkhals is not installed properly, a lot of files are missing. You might have interrupted the installation somehow.

That is what I thought but I didn't interrupt the installation.  It rebooted after install on it's own.  I'll give it another try.

kwe1009, 2025-03-08:
> [@kwe1009](https://github.com/kwe1009) Rinkhals is not installed properly, a lot of files are missing. You might have interrupted the installation somehow.

Just tried to reinstall:
1. installed thumb drive
2. heard one beep after a few seconds
3. progress bar showed installing and then single beep
4. rebooted on it's own had heard 2 quick beeps.

here is the install log (not sure why the date/time is messed up

[debug-bundle_3907738554_20250309-050647.zip](https://github.com/user-attachments/files/19145905/debug-bundle_3907738554_20250309-050647.zip)

:
`Sun Mar  9 04:57:39 CST 2025: 
Sun Mar  9 04:57:39 CST 2025: Starting Rinkhals installation...
Sun Mar  9 04:57:40 CST 2025: Backing up machine-specific files
Sun Mar  9 04:57:40 CST 2025: Installing Rinkhals version 20250308_01
Sun Mar  9 04:57:40 CST 2025: Copying Rinkhals files
Sun Mar  9 04:58:10 CST 2025: 
Sun Mar  9 04:58:10 CST 2025: Starting Rinkhals installation...
Sun Mar  9 04:58:11 CST 2025: Backing up machine-specific files
Sun Mar  9 04:58:12 CST 2025: Installing Rinkhals version 20250308_01
Sun Mar  9 04:58:12 CST 2025: Copying Rinkhals files
Sun Mar  9 04:58:15 CST 2025: Copying Rinkhals startup files
Sun Mar  9 04:58:15 CST 2025: Rinkhals loader was detected, skipping installation
Sun Mar  9 04:58:15 CST 2025: Removing update files
Sun Mar  9 04:58:18 CST 2025: Copying Rinkhals startup files
Sun Mar  9 04:58:18 CST 2025: Rinkhals loader was detected, skipping installation
Sun Mar  9 04:58:18 CST 2025: Removing update files
Sun Mar  9 04:58:19 CST 2025: Rinkhals installation complete, rebooting...
Sun Mar  9 04:58:19 CST 2025: Rinkhals installation complete, rebooting...
`

jbatonnet, 2025-03-09:
That's crazy, you're experiencing a normal install, but files are missing in the debug bundle.
You now have more files, but you're still missing some important ones, like `/useremain/rinkhals/start-rinkhals.sh` to actually start Rinkhals.

A typical Rinkhals install is around 250 MB, your logs show 49.3 MB for 20250308_01...

The only weird thing I can see here is that the `Rinkhals installation complete, rebooting` is listed twice.
I also see in your active processes that channel.sh, update.sh and more are there twice.
```
 2737  2721 root     S     1224   1%   0% {channel.sh} /bin/sh /userdata/channel
 2738  2724 root     S     1224   1%   0% {channel.sh} /bin/sh /userdata/channel
 2802  2800 root     S     1220   1%   0% {update.sh} /bin/sh /useremain/update_
 2798  2797 root     S     1220   1%   0% {update.sh} /bin/sh /useremain/update_
 2800  2741 root     S     1216   1%   0% /bin/sh -c /useremain/update_swu/updat
 2797  2742 root     S     1216   1%   0% /bin/sh -c /useremain/update_swu/updat
```

There might be something weird going on here, and I suspect something not related to Rinkhals.
Did anything weird happen during some stock firmware installation or update?

jbatonnet, 2025-03-10:
Did you try the other USB slot by any chance?

kwe1009, 2025-03-10:
I just did another system reset on the K3 and tried to install Rinkhals again, this time in the left USB slot.  Still got the double beep after reboot.  

[debug-bundle_3907738554_20250311-033937.zip](https://github.com/user-attachments/files/19169701/debug-bundle_3907738554_20250311-033937.zip)

kwe1009, 2025-03-11:
@jbatonnet I reset back to 2.3.3.9, updated to 2.3.5.3 and then 2.3.7.1.  I also did the config reset you made and still no love getting Rinkhals installed.  I unplugged the ACE and used the left USB slot.

[debug-bundle_3907738554_20250312-052625.zip](https://github.com/user-attachments/files/19198286/debug-bundle_3907738554_20250312-052625.zip)

jbatonnet, 2025-03-12:
Hi @kwe1009, thanks for your help on collecting those logs.
Unfortunately, it confirms what I thought:

Here is a normal debug bundle / ps.log (confirmed on multiple printers / users):
[image]

And here is yours:
[image]

They are all running twice. It means some system files are different for some reason. I guess udev is configured to start twice somehow, and it persists across firmwares.
Without spending a long time diagnosing that, I don't really see what can be done. Even if I find tricks short term, it will bite at some point.

I wrote some internal file system documentation here: https://github.com/jbatonnet/Rinkhals/wiki/Reverse-Engineering-%E2%80%90-File-structure#kobra-startup-sequence

You might want to check if any files are corrupted somehow. I guess somewhere around `/lib/udev/rules.d/61-udisk-auto-mount.rules`

jbatonnet, 2025-03-12:
Oh and maybe an idea, if your USB drive has multiple partitions you may have issues too, I don't know.

closed 2025-03-19 by jbatonnet