jbatonnet/Rinkhals #22: Request: a way to safe your z offset

ducati1198, 2025-01-08:
A way to safe z offset in fluid or mainsail to be able to start the print with your custom z offset without putting any gcode in the slicer machine start gcode. The only way i was able to make my print start with my z offset automatic is by creating a gcode call MYOFFSET and put it in the slicer. Again thanks and appreciate rinkhals

Yangyuanxin, 2025-02-13:
I don't think this is the best way to solve this problem. Firstly, you need to ensure that your nozzle is clean enough, as the accuracy of the zoffset is largely influenced by it. In addition, uneven hot bed, mismatched hot bed temperature, pressure sensor failure, and unsmooth Z-axis screw can also greatly affect it.

ducati1198, 2025-02-13:
> I don't think this is the best way to solve this problem. Firstly, you need to ensure that your nozzle is clean enough, as the accuracy of the zoffset is largely influenced by it. In addition, [...]

Everything you mention as been done and only wait to have a perfect layer was by adjusting the z offset. Kobra 3 bed are extremely cheap and warps a lot. To fix the warping issue I added aluminum tape between the bed and the belt plate in low spots. I didn’t find any other wait to adjust the z offset, Anycubic always takes over before the print

closed 2025-03-08 by jbatonnet