jbatonnet/Rinkhals #23: Printprogress and corresponsing features on kobra screen

EikeKre, 2025-01-10:
Hey, I am wondering if it is implemented that the first tab on the kobra screen shows the current progress when starting a print via webinterface/orcaslicer? 
On my maschine it shows just the print button as if there is nothing running.

jbatonnet, 2025-01-10:
Hi, yes it is implemented. Make sure you have LAN mode enabled on your printer.
Then if Rinkhals has started (when you have access to SSH and Mainsail for example), any print from anywhere will trigger the print screen.

Schrolli91, 2025-03-15:
This is not the case with my Kobra S1 on `20250311_01`.
If i start a print from the web interface or trough Orca, the Printer Screen still remains in the menu.

jbatonnet, 2025-03-17:
Next release will fix this issue

jbatonnet, 2025-03-17:
Should be working as of https://github.com/jbatonnet/Rinkhals/releases/tag/20250317_01

closed 2025-03-17 by jbatonnet

Schrolli91, 2025-03-17:
I'll try that tomorrow afternoon - Thanks for your efforts!