jbatonnet/Rinkhals #24: Printing from Webinterface not working?

NatroNx, 2025-01-13:
The webinterface works  just fine and printing via Anycubic Slicer Next works as well. But I cannot print from within the webinterface via choosing a file and hit print.  Just nothing happens.

Is this working for you?

jbatonnet, 2025-01-14:
Yes it is supposed to be working. 
This firmware includes a proxy that converts print calls to use the default print behavior this triggering the print screen. 
This feature is only supported using LAN mode. If it is not enabled, the proxy doesn't interfere with the print. 
Can you check if LAN mode is enabled or disabled on your printer, and generate a debug bundle (swu tools instructions: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools)

NatroNx, 2025-01-14:
Hello and thanks for the quick response. 
I got the support bundle but can't find much in the files. Maybe you can have a log. 
[debug-bundle_2819455134_20250115-041713.zip](https://github.com/user-attachments/files/18415831/debug-bundle_2819455134_20250115-041713.zip)

jbatonnet, 2025-01-14:
Thank you, I see one error about the proxy while trying to print:

```
2025-01-14 20:16:51 [moonraker-proxy] Intercepted "printer.print.start", replacing with MQTT call...
/usr/share/scripts/moonraker-proxy.py:172: RuntimeWarning: coroutine 'mqtt_printfile' was never awaited
  mqtt_printfile(data['params']['filename'])
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

This is indeed a mistake on my side and I'm going to release a patched version soon.
In the meantime, I see that the mistake is not there if you try to print directly from Orca. You can try that instead until next release!

jbatonnet, 2025-01-14:
I can confirm I could repro with 20250107, but it is now fixed with [20250114_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250114_01)

closed 2025-01-14 by jbatonnet

NatroNx, 2025-01-14:
confirm it is fixed!