jbatonnet/Rinkhals #25: cuenta no encontrada

yamahac70, 2025-01-14:
[image]
firmware stock Kobra3_2.3.5.3

yamahac70, 2025-01-14:
Sorry, I forgot to say that I installed version 20250107_01 of Rinkhals

jbatonnet, 2025-01-14:
Oh okay, then I'm reopening this issue, do you have LAN mode enabled?
I admit I'm only using my printer in LAN mode now so the Anycubic account feature hasn't been tested properly. 
The firmware doesn't really change much in this regards, but it might be something i missed. 
Can you generate a debug bundle of your printer to see if there's anything in the logs ? The instructions on how to use swu tools are there: 
https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

closed 2025-03-08 by jbatonnet