jbatonnet/Rinkhals #27: Help to fix please: Spoolman docker needs to get restart to start reading the filament from the printer

ducati1198, 2025-01-15:
@jbatonnet  sorry im not sure if it the right place to ask your your help. Iwas able to get pool man working on the goklipper by modifying spoolman.py, like mantion in https://github.com/utkabobr/DuckPro-Kobra3/issues/54 but not i always need to restart spoolman docker to start reading the filament. i get this errot in moonraker log. im not able to find the problem, i try difference router and pc with spoolman installed on it and always the same problem. with my creality every thing eork and theres no error in log
sorry if this not the right place to ask you can erase my post

Anycubic 10.11.88.40 issue
Creality 10.11.88.30 Everything works good

[moonraker.log](https://github.com/user-attachments/files/18427753/moonraker.log)
[image]
[image]

jbatonnet, 2025-01-15:
For spoolman I'll need to add support, but I'm looking at a general solution. This thing is there's an issue in gklib where it puts everything in uppercase, and discard some gcode for some reason. I'd love to be able to patch the binary or even better rebuild it (because we have the source, but not the latest version) with those fixes.
Progress will be discussed here: https://github.com/jbatonnet/Rinkhals/issues/1

Now, for your issue, since Spoolman only exists in Moonraker and interacts with a separate server, I really feel like my proxy cannot do much. But for testing, can you connect your Mainsail/Fluidd to <PRINTER_IP>:7126 instead of 7125.
7126 exposes the raw Moonraker connection without any Rinkhals stuff, and 7125 is the proxy.
If you have the same behavior with 7126, you should try to get some help from Spoolman or Moonraker folks

In your moonraker log, I can see a LOT of those lines. Might be related?
```
[announcements.py:_fetch_moonlight()] - Failed to update subscription 'moonraker': [Errno 97] Address family not supported by protocol
[announcements.py:_fetch_moonlight()] - Failed to update subscription 'klipper': [Errno 97] Address family not supported by protocol
[announcements.py:_fetch_moonlight()] - Failed to update subscription 'spoolman': [Errno 97] Address family not supported by protocol
```

And some of those as well:
```
[spoolman.py:report_extrusion()] - Spool ID 0 not found, setting to None
[base_events.py:default_exception_handler()] - Task exception was never retrieved
future: <Task finished name='Task-882052' coro=<FlexTimer._call_wrapper() done, defined at /usr/share/moonraker/moonraker/eventloop.py:208> exception=TypeError("int() argument must be a string, a bytes-like object or a real number, not 'NoneType'")>
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/eventloop.py", line 214, in _call_wrapper
    ret = await ret
          ^^^^^^^^^
  File "/usr/share/moonraker/moonraker/components/spoolman.py", line 319, in report_extrusion
    self.set_active_spool(None)
  File "/usr/share/moonraker/moonraker/components/spoolman.py", line 282, in set_active_spool
    spool_id = int(SPOOL_ID)
               ^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

Again, this would not be related to Rinkhals, but maybe those clues might help?

ducati1198, 2025-01-15:
hen you say connect to 7126 instead of 7125 is in orca slicer ?

jbatonnet, 2025-01-15:
Orca uses the octoprint_compat plugin from Moonraker, so it won't work for it.
But when you are using Mainsail, Fluidd or any other web client you're using to access the printer and Spoolman, you might be able to specify the port.

For example in Fluidd:
[image]

But saying that, Spoolman has mothing to do with the web client, I really don't think it could make a difference.

jbatonnet, 2025-01-20:
To keep the backlog tidy I'll merge this one with Spoolman support

closed 2025-01-20 by jbatonnet