jbatonnet/Rinkhals #29: I don't fill in device_account.json correctly.

Zloyfey, 2025-01-25:
Good day! Many thanks for your work!
I am trying to restore the board after experiments. When I used copies of my Userdata b Useremain everything worked. After trying to install DuckPro it hangs on the logo and I can't get it back. I can't do this from the instructions on how to restore userdata

/userdata/app/gk/config/device_account.json
deviceId, username and password should be filled properly
deviceId must be the same as /useremain/dev/device_id

Questions:
1. What happens if I don't fill in device_account.json correctly? Can it hang on the logo?
2. In /useremain/dev/ I only have mashine_id. Where can I try to find device_id?
3. In device_account.json "username": "xxx","password": "xxx" - is this the login and password for the anycubic account?
4. In device_account.json "devicecrt": "xxx","devicepk": "xxx", I don't even roughly understand where to find it, what topic to study to understand this...

jbatonnet, 2025-01-25:
Hi @Zloyfey, thank you for your message.
Some answers:

1. I don't think this would hang but you might see weird behaviors (Anycubic Slicer won't connect, prints not starting properly, ...)
2. 
3. If you don't have any backup of your device id, I don't know how to get it back. If you tried to install Rinkhals at least once, a backup is automatically created on the USB driveI personnaly have those files in /useremain/dev:
[image]

I uploaded partitions backup over there: https://1drv.ms/f/s!Ag-0E4JXrqAlwbwtxJkvEgvxqJbI4Q?e=vYBveG
It contains all the files, but the deviceId / other personnal info are missing.
It might be enough to unblock you, those are the files I use when I brick my printer, and it starts without the personnal data.

3. No those are the credentials for the embedded Mochi MQTT server. They are used to exchange information between Anycubic apps, and by my proxy to trigger the print screen when launching a print from Orca.

4. Yeah, I'm not sure to fully understand the certificate management on the Kobra 3 yet. I know there are 2 ports to access MQTT, 2883 and 9883. The 9883 is "secure", this one might use those certs?

jbatonnet, 2025-01-25:
Great, I'm glad the instruction could help you!

closed 2025-01-26 by jbatonnet