jbatonnet/Rinkhals #33: MQTT Connectivity

kwe1009, 2025-02-05:
I am trying to connect to MQTT using the instructions at https://github.com/jbatonnet/Rinkhals/wiki/Reverse-Engineering-%E2%80%90-MQTT but can't get MQTT Explorer to connect.  I just get a "disconnected from server" message.  I use MQTT quite a bit to port things into Home Assistant and would like to do it with the Kobra 3 also.  Any suggestions?  Thanks.

jbatonnet, 2025-02-05:
Hi @kwe1009, thank you for your message. You will usually get that message when LAN mode is not enabled. From what I gathered, the internal MQTT server is only started when LAN mode is enabled, during Anycubic services startup.
If it's already enabled, you should connect using SSH and check what ports are open using `netstat -tln` in order to validate that ports 2883 and 9883 are open.

Then I'm thinking of your objective connecting the Kobra to Home Assistant. I wrote a simple Go app to export some system metrics to a MQTT broker, and I added the Home Assistant device discovery message to it's easily recognized as a couple of sensors.
I made it available as an app in Rinkhals.

Documentation is here: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-monitor
Source is here: https://github.com/jbatonnet/Rinkhals/blob/master/files/4-rinkhals/home/rinkhals/apps/rinkhals-monitor/rinkhals-monitor.go

kwe1009, 2025-02-06:
Thanks!  I wasn't in LAN mode but now that I am it still doesn't work.  Here is the output from netstat:

[image]

I was looking at the app previously but can't get over this hurdle unfortunately.

jbatonnet, 2025-02-07:
Ports 2883 and 9883 are open, you should be able to connect using a MQTT client now.
The page https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-monitor explains how to find username and password.
Port 2883 doesn't use TLS. I'm using MQTT Explorer to connect and it works.
If you still cannot connect using your own client, it's most likely a client issue.

closed 2025-02-08 by jbatonnet