jbatonnet/Rinkhals #36: Nginx crashes when uploading files above a certain size

jbatonnet, 2025-02-10:
**Describe the bug**
Nginx crashes when uploading files above a certain size

**To Reproduce**
Steps to reproduce the behavior:
1. Upload a big gcode file, for example 117MB

w1ll14m, 2025-02-10:
Ahh I ran into this problem too. a 40MB file was enough to crash it in my case initially I thought I had an issue with my gcode

jbatonnet, 2025-02-10:
Okay, I fixed the proxy by now streaming the gcode to Moonraker.
The proxy by itself is not an issue anymore and I can load / print 100MB files from Orca.

But some other processes - out of my control - still reserve memory for the size of the file, and it still crashes around ~120MB.
For files this big, I would advise copying the file on the printer or using a USB drive for now.

The proxy fix will be in next release

jbatonnet, 2025-02-10:
Should be fixed in [20250210_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250210_01)

closed 2025-02-10 by jbatonnet