jbatonnet/Rinkhals #43: [BUG]Rinkhals 20250217_01: When activating Rinkhals monitor the system get freeze

rabbit3dcustom, 2025-02-18:
**Describe the bug**
Printer: Kobra 3 (+ combo) with firmware 2.3.5.3

The Rinkhals cfw install sucessfully and everything is working perfectly, orcaslicer is sending files, print it correctly.

But, when i decided to start using the mqtt capabilites and activate the Rinkhals monitor 0.1, that is already in the cfw, when pressed the switch,  the printer get stuck, freeze... no respond.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on printer
2. Go to settings/Rinkhals
3. Click on Manage Apps
4. Click on Activate Rinkhals monitor(0.1)
5. Printer freezes
6. 

It happends with the three applications, none of them is working.

I downgrade to version 20250210_01 and same happended.

**Logs / Additional information**
Use the debug bundle SWU tool to collect logs on your printer before posting a new issue.
Instructionas are available here: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

I'm trying to install the SWU tools, but it seems not be working, not installing.

Kind regardas and thank you so much in advance.
Rabbit

jbatonnet, 2025-02-18:
Hi @rabbit3dcustom, thank you for your message.

I got this report twice already, I'm trying to get a repro on my printer.
Even if this is additional optional features, I really want to have a stable app system so anyone can add features.

rabbit3dcustom, 2025-02-18:
Hi jbtonnet,

Apologies, was trying to find any kind of similar issue, but i do not found it. 

There is no problem if you delete this post, i don't want to bother you.

Right now, i'm very glad and happy to be able to send the gcodes via Orcaslicer instead of the Anycubic Frankenstein-slicer XD

Kind regards
Rabbit.

jbatonnet, 2025-02-18:
Ok found the bug, a fix is on the way!

jbatonnet, 2025-02-18:
[20250218_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250218_01) fixes the issue, thank you again!

closed 2025-02-18 by jbatonnet

rabbit3dcustom, 2025-02-19:
Thanks a lot!!!!
Damm you were fast!!!!