jbatonnet/Rinkhals #45: Code 11407 after reboot latest release

Lecter224, 2025-02-22:
After flashing and using latest firmware Rinkhals 20250221_01 I get this morning code 11407 after fresh restart. 

Your wiki doesn't helped me out, it was not possible to recover with stock firmware 2.3.3.9, I have to use latest one 2.3.5.3 to get my kobra to work.

All other releases worked perfect!

jbatonnet, 2025-02-22:
That's weird, Rinkhals shouldn't be able to trigger this 11407 by itself. Did you install anything that could have modified the printer.cfg file or by yourself at some point?
Everytime I had a similar issue I was able to restore stock 2.3.3.9, factory reset and then I usually do the official OTA to 2.3.5.3. Did you try that?

Lecter224, 2025-02-22:
Maybe the DuckPro firmware weeks before, but deleted it same way with stock firmware and master reset like yours. I used your firmware for about two weeks, every day printing. Then I did update yesterday and it worked for hours. This morning I restart machine and get this error. I was not able to restore with 2.3.3.9, three times and two downloads i test, but evertime it gave the normal sound, I wait for 30min, restart machine, error 11407 again. So I used 2.3.5.3 and it works at first try. But wait, yesterday I tried for the first time your menue on printer, started and stopped your features. Maybe there is a hint...

jbatonnet, 2025-02-22:
Okay, so something you can try is to disable Rinkhals manually on startup. Create a file named `.disable-rinkhals` at the root of a USB key, plug it in the printer and restart. During boot, Rinkhals will not start and you'll get back to stock.
If you still get 11407, there's definitely something else causing the error. Then you may have to play with the factory reset to get it back on track.

Lecter224, 2025-02-22:
At the moment I am on stock because I was not sure your firmware has a little bug. Next days I will update again to your firmware and if I get an error again, i will test the file on usb- stick to disable ringkhals.

jbatonnet, 2025-02-22:
That's fair. I believe there's something else cause many people use the firmware without issues, let me know what happens then.

rabbit3dcustom, 2025-02-23:
Hi, Something like this happened to me, raise and error that a original firmware was not present and the printer could be damaged.

But, nothing to worry, I turned off the printer, wait a few minutes and then the Rinkhals started correctly without any issue.

Lecter224, 2025-03-01:
After flashing latest release Rinkhals 20250223_01 I get the same error 1140. With the .disble-file on root of usb-stick, printer starts normal. Without I get always this error. I Think there is something wrong on my stock firmware. How can I cleanup the hole machine to begin with clean conditions? Flashing latest stock is not the solution I think...

jbatonnet, 2025-03-08:
Hi, if the printer starts well with the disable file, you might have a corrupted configuration. Try the latest release, or use the SSH SWU tool to access SSH, and `rm -rf /useremain/home/rinkhals` to reset Rinkhals to default settings

jbatonnet, 2025-03-12:
I wrote some additional documentation here about 11407: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Printer-configuration

closed 2025-03-12 by jbatonnet