jbatonnet/Rinkhals #46: [question] Anycubic Kobra S1 support

Ziomal12, 2025-02-22:
Hello,
I'm thinking of getting S1 and I have two questions.
1. Is the support for S1 likely to be added?
2. Does AcePro function as in the stock firmware?
3. Optional, but is it possible to install tailscale or at least wireguard on this firmware?

Thank you for your great work!

jbatonnet, 2025-02-22:
Hi @Ziomal12, here are some answers:
1. Yes I'd like to add it. I don't know for sure, but I believe Anycubic used similar components making it possible to use the same or a similar firmware. I don't personally have the S1, but I'm looking for some people to see what can be done
2. Rinkhals doesn't change or interfere with the default features, so yes the ACE Pro will still work
3. The main challenge on the Kobra printers is the limited performance in terms of CPU and Memory. I have a prototype of Tailscale running on the printer but didn't try using it a lot. But yeah it would be interesting to see it running properly: https://github.com/jbatonnet/Rinkhals.apps/tree/feature/tailscale/apps/tailscale
Wireguard would be more complex because the kernel doesn't support low level networking / tunneling. Tailscale has a userspace networking system.

Ziomal12, 2025-02-22:
Thank you very much. If you give me some commands I'll dump as much info as possible when I get it (hopefully in late March/April).

jbatonnet, 2025-02-22:
Thank you that would be helpful, please join the Discord when you get it!
I'm closing this issue, but feel free to ask more question or hop on Discord if you want more info.

closed 2025-02-22 by jbatonnet