jbatonnet/Rinkhals #48: Kobra3 latest FW 2.3.7 incompatibility.

Art531, 2025-03-03:
**Describe the bug**
After flashing Kobra3 latest firmware 2.3.7 Rinkhals custom cannot be installed any more.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash latest official firmware
2. Try to flash latest Rinkhals custom
3. Flashing finishes with red line and 3 beeps

**Logs / Additional information**
Installation log on usb stick says that ершу Rinkhals release incompatible with Kobra firmware 2.3.7

lequim2010, 2025-03-06:
Jbatonnet is working on a new version for the 2.3.7, I think you will just have to wait

christophervankammen, 2025-03-07:
I would love to offer any help with this if needed! This work makes this printer actually something worthwhile!

jbatonnet, 2025-03-07:
@christophervankammen if you're interested you can join the Discord listed on the home page where we discuss the progress / test and other things

jbatonnet, 2025-03-07:
Next release will include compatibility with Kobra 3 with 2.3.7.1 fw. I will skip 2.3.7

jbatonnet, 2025-03-08:
Released in https://github.com/jbatonnet/Rinkhals/releases/tag/20250308_01

closed 2025-03-08 by jbatonnet

cdvankammen, 2025-03-09:
bless you for your work!