jbatonnet/Rinkhals #50: Kobra S1 - Prints in Orca Slicer won't automatically start from "Preview" - "Print" but will from uploading gcode from Device

Ellevix, 2025-03-08:
In Orca slicer, after having sliced the build and hitting print from the "Preview" tab, I select what to name the G-Code then "Upload and Print". The file attempts to send to the printer however then returns an error: "

Sometimes the actual file does upload onto the printer however the print job fails to start automatically.
However if I save the g-code to a local drive, then click "Device"  the print will upload and start without any issues.

Steps:
1. After slicing the build I click "Print" and I'm presented with the "Send G-Code to printer host" screen

[image]

2. Give the g-code file a name
3. Click "Upload and Print"
4. In the lower right corner the status is reported as "g-code name -> ip address COMPLETED" however then changes to "g-code name -> ip address ERROR"
5. An error is then presented "Error uploading to print host: HTTP 400: [moonraker-proxy] Timeout trying to print Base_PLA_0.12_1h8m.gcode" and the print does not automatically start printing.

[image]

6. Click ok on the error
7. On the preview screen click "Export G-Code File"
8. Save the g-code file to a local drive
9. In Orca Slicer click the "Device" tab
10. Click "Upload & Print"

[image]

11. Select the local g-code file and click ok
12. The files then uploads and print automatically starts.

A clear and concise description of what the bug is, and what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Logs / Additional information**
Use the debug bundle SWU tool to collect logs on your printer before posting a new issue.
Instructionas are available here: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#swu-tools

jbatonnet, 2025-03-12:
Hi @Ellevix, thank you for your message.
The two moonraker warning are captured in another issue, they do not impact functionalities.
I'm aware of some issues about moonraker-proxy, but it's difficult to diagnose since I don't have the S1 myself.

In the meantime, I made the proxy as an app so it's easy to enable / disable and get things working. It's available in last release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250311_01

To disable an app, you need to connect using SSH, then `touch /useremain/home/rinkhals/apps/39-moonraker-proxy.disabled`.
Then you can reboot: `sync && reboot`

Ellevix, 2025-03-16:
@jbatonnet, thank you for the information and suggestion to disable Moonraker, I haven't tried it yet but will do so and let you know.
I'm happy to help you out with any testing for the S1 for this and any other issues/improvements, as long as it doesn't brick my machine :)

jbatonnet, 2025-03-17:
This will be fixed in next realease

jbatonnet, 2025-03-17:
It should now be working as of https://github.com/jbatonnet/Rinkhals/releases/tag/20250317_01