jbatonnet/Rinkhals #52: After Change the printer.custom.cfg Failure 11407

Joerg1309, 2025-03-10:
I have changed the printer.custom.cfg and wanted to set the mash level to 8.8 which is set to 5x5 by default. after the change the printer comes up with the error 11407.
Rinkhals is deactivated and the Rinkhals item is no longer visible on the display. 

Installing Rinkhals again was unsuccessful and the same error occurred.

Installing the original 2.3.7.1 again and then reverting to Rinkhals did not result in Rinkhals running again. A message appears that there is no original firmware on the device but Rinkhals will not start.

jbatonnet, 2025-03-12:
I wrote some documentation here: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Printer-configuration

closed 2025-03-12 by jbatonnet

KarimSouilmi, 2025-04-14:
Hi,

I also tried to increase the bed mesh. Where can I find the .enable-rinkhals file? Or how can I create it?

Only the printer.custom.cfg on the root of the USB doesn't work. 

Kind regards,

Karim

jbatonnet, 2025-04-14:
`.enable-rinkhals` can be created at the root of the USB drive

KarimSouilmi, 2025-04-14:
just as a text file?

jbatonnet, 2025-04-14:
An empty file, only the name and the location are important

KarimSouilmi, 2025-04-14:
thank you verry much. I'm I correct I don't have the rinkhals icon in the settings on a ks1?