jbatonnet/Rinkhals #55: Kobra 3 Display freeze and now connnection to Klipper via Web Gui and Cam with now piture

gelberlemmy, 2025-03-11:
I have installed the actually Rinkhals-Firmware to my cobra 3. First of all i have make a clean orginal Anycubic Firmware on the mashine. (before i have DuckPro on the maschine. But this firmware is very buggy)

I go to the defaults and then i will install Rinkhals. The installation was successfull. I can go on the Klipper WebGui.
Firts of all i have two massages with warnings. See attachmend. 
And a little bit later my Touchdisplay will freeze und no connection via web gui. 
I have downlaodet the logs via FireZilla. This connection still works. 

What can i do. I go back again to orginal Firmware and than Rinkhals. Same Bug.

Regards Andre

[image]

[monitor.log](https://github.com/user-attachments/files/19197728/monitor.log)
[rinkhals.log](https://github.com/user-attachments/files/19197727/rinkhals.log)

jbatonnet, 2025-03-12:
Link is here: https://github.com/jbatonnet/Rinkhals/actions/runs/13793952378

gelberlemmy, 2025-03-12:
Good morning, have you corrected something since yesterday ? i see a new Firmware File

jbatonnet, 2025-03-12:
Unfortunately the modifications are not related to this issue

gelberlemmy, 2025-03-12:
Okay, how can I send you the results ?

jbatonnet, 2025-03-12:
Here on Github, or on the Discord server linked on the main page

gelberlemmy, 2025-03-12:
Okay. Actually I have restored it to 2.3.5.3. than OTA to 2.3.7.1. set it back to factory. Than the Rinkals Firmware. It runs for a moment. Than I have a crash from the Klipper Web GUI. Than now the WLan are gone and comes later back and no connection to Klipper anymore.  But the Display don’t freeze after Updating.

[debug-bundle_83563934_19700101-000114.zip](https://github.com/user-attachments/files/19218303/debug-bundle_83563934_19700101-000114.zip)

jbatonnet, 2025-03-19:
Newer versions of Rinkhals should improve things considerably for this kind of issues. Can you check again with newer versions?

gelberlemmy, 2025-03-23:
Good Morning, my son updatet the Printer to 2.3.8. So would the Firmware work on it ? So i can test it on 2.3.8

jbatonnet, 2025-03-23:
Morning, yes 2.3.8 is supported

closed 2025-03-23 by jbatonnet

gelberlemmy, 2025-03-27:
Okay, I have testet it. And for the first Print it works perfectly. Thanks for your support. When I find something else I will post it