jbatonnet/Rinkhals #62: Missing sensors in HA moonraker integration

Hinguy, 2025-03-16:
Hello,
I got the moonraker custom integration in home assistant working with the custom firmware on my Kobra S1. It is showing all the buttons to control the printer, however all the sensors are missing, so I can't integrate them into home assistant to do automations. Is anyone also using home assistant and got this working?

I don't know if it's a problem of the moon raker integration in HA or a "problem" in the firmware, because it is a modified version of Klipper. Unfortunately I am not experienced enough, so I don't know where to begin with troubleshooting.

Thanks for the help, well appreciated! 👍

DerGary, 2025-03-17:
@Hinguy see: https://github.com/marcolivierarsenault/moonraker-home-assistant/issues/465

Hinguy, 2025-03-17:
> [@Hinguy](https://github.com/Hinguy) see: [marcolivierarsenault/moonraker-home-assistant#465](https://github.com/marcolivierarsenault/moonraker-home-assistant/issues/465)

@DerGary oh thank you very much, will have an eye on that 👍

DerGary, 2025-03-26:
Upgraded to Rinkhals 20250325_02, but the issue still persists

MichiMess, 2025-03-31:
Yeah, I also missing the sensors

jbatonnet, 2025-04-16:
I recently added a lot of things to make Mainsail, Fluidd and OctoApp working well with Rinkhals. I think we reached a point where Rinkhals Moonraker behaves well enough so other clients may need to improve a little bit their resiliency to Klipper variants.

If there is something "simple" I can fix to improve this further tell me, else this could be on the HA integration side.
Let me know