jbatonnet/Rinkhals #65: Printer restarts mid print

DerGary, 2025-03-17:
**Describe the bug**
After latest update can't print anymore as the printer restarts mid print due to some error.

**To Reproduce**
Start a print and wait. Happened to me twice in two prints in 5 minutes.

**Logs / Additional information**

Haven't tried the debug tools but here is the moonraker log, is this enough?:

```
2025-03-17 08:15:14,015 [server.py:load_component()] - Component (shell_command) loaded
2025-03-17 08:15:14,016 [machine.py:__init__()] - Using System Provider: none
2025-03-17 08:15:14,052 [server.py:add_log_rollover_item()] - Module libcamera unavailble, import failed
2025-03-17 08:15:14,052 [server.py:load_component()] - Component (machine) loaded
2025-03-17 08:15:14,067 [server.py:load_component()] - Component (data_store) loaded
2025-03-17 08:15:14,076 [proc_stats.py:__init__()] - Unable to find 'vcgencmd', throttle checking disabled
2025-03-17 08:15:14,081 [server.py:load_component()] - Component (proc_stats) loaded
2025-03-17 08:15:14,092 [server.py:load_component()] - Component (job_state) loaded
2025-03-17 08:15:14,147 [server.py:load_component()] - Component (job_queue) loaded
2025-03-17 08:15:14,176 [database.py:register_table()] - Found registered table job_history
2025-03-17 08:15:14,178 [database.py:register_table()] - Found registered table job_totals
2025-03-17 08:15:14,190 [server.py:load_component()] - Component (history) loaded
2025-03-17 08:15:14,221 [server.py:load_component()] - Component (http_client) loaded
2025-03-17 08:15:14,247 [server.py:load_component()] - Component (announcements) loaded
2025-03-17 08:15:14,263 [server.py:load_component()] - Component (webcam) loaded
2025-03-17 08:15:14,277 [server.py:load_component()] - Component (extensions) loaded
2025-03-17 08:15:14,316 [server.py:load_component()] - Component (octoprint_compat) loaded
2025-03-17 08:15:14,329 [server.py:_initialize_component()] - Performing Component Post Init: [dbus_manager]
2025-03-17 08:15:14,332 [dbus_manager.py:component_init()] - Unable to Connect to D-Bus
2025-03-17 08:15:14,333 [server.py:_initialize_component()] - Performing Component Post Init: [database]
2025-03-17 08:15:14,348 [server.py:_initialize_component()] - Performing Component Post Init: [file_manager]
2025-03-17 08:15:15,081 [server.py:_initialize_component()] - Performing Component Post Init: [authorization]
2025-03-17 08:15:15,144 [server.py:_initialize_component()] - Performing Component Post Init: [machine]
2025-03-17 08:15:15,161 [machine.py:validation_init()] - Installation version in database up to date
2025-03-17 08:15:15,167 [server.py:_initialize_component()] - Performing Component Post Init: [proc_stats]
2025-03-17 08:15:15,172 [server.py:_initialize_component()] - Performing Component Post Init: [history]
2025-03-17 08:15:15,210 [server.py:_initialize_component()] - Performing Component Post Init: [announcements]
2025-03-17 08:15:15,227 [server.py:_initialize_component()] - Performing Component Post Init: [webcam]
2025-03-17 08:15:15,244 [server.py:_initialize_component()] - Performing Component Post Init: [klippy_connection]
2025-03-17 08:15:15,267 [extensions.py:start_unix_server()] - Creating Unix Domain Socket at '/userdata/app/gk/printer_data/comms/moonraker.sock'
2025-03-17 08:15:15,271 [server.py:start_server()] - Starting Moonraker on (0.0.0.0, 7126), Hostname: Rockchip
2025-03-17 08:15:15,275 [application.py:listen()] - SSL Certificate/Key not configured, aborting HTTPS Server startup
2025-03-17 08:15:15,533 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2025-03-17 08:15:15,797 [server.py:add_warning()] - Supplied path (/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/config) for (config_examples) is invalid. Make sure
that the path exists and is not the file system root.
2025-03-17 08:15:15,799 [server.py:add_warning()] - Supplied path (/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/docs) for (docs) is invalid. Make sure
that the path exists and is not the file system root.
2025-03-17 08:15:16,127 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2025-03-17 08:15:16,139 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2025-03-17 08:15:16,377 [job_state.py:_handle_started()] - Job state initialized: standby
2025-03-17 08:15:16,635 [klippy_connection.py:_verify_klippy_requirements()] - Configuration for [virtual_sdcard] not found, unable to set SD Card path
2025-03-17 08:15:16,659 [klippy_connection.py:_check_ready()] - Klippy ready
2025-03-17 08:15:17,139 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'extruder']
2025-03-17 08:15:17,302 [announcements.py:_fetch_moonlight()] - Failed to update subscription 'moonraker': [Errno -2] Name or service not known
2025-03-17 08:15:19,350 [announcements.py:_fetch_moonlight()] - Failed to update subscription 'klipper': [Errno -2] Name or service not known
2025-03-17 08:15:23,042 [job_state.py:_status_update()] - Job Started: Deckel_PLA_0.2_41m41s.gcode
2025-03-17 08:15:23,103 [server.py:_process_event()] - 
Error processing callback in event job_state:state_changed
Traceback (most recent call last):
  File "/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/history.py", line 379, in _on_job_state_changed
    self.job_paused = job_event == JobEvent.PAUSED
                                   ^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250316_02/usr/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: PAUSED

2025-03-17 08:16:06,463 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 172.16.3.2
2025-03-17 08:16:06,469 [application.py:log_request()] - 101 GET /websocket (172.16.3.2) [_TRUSTED_USER_] 11.40ms
2025-03-17 08:16:06,471 [websockets.py:open()] - Websocket Opened: ID: 2763855024, Proxied: False, User Agent: HomeAssistant/2025.3.3 aiohttp/3.11.13 Python/3.13, Host Name: 172.16.3.28
[... 20 lines cut ...]
2025-03-17 08:19:41,871 [server.py:_process_event()] - 
Error processing callback in event job_state:state_changed
Traceback (most recent call last):
  File "/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/history.py", line 379, in _on_job_state_changed
    self.job_paused = job_event == JobEvent.PAUSED
                                   ^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250316_02/usr/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: PAUSED

2025-03-17 08:19:42,866 [eventloop.py:_wrapper()] - Error Running Callback
Traceback (most recent call last):
  File "/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/moonraker/moonraker/eventloop.py", line 89, in _wrapper
    await ret
  File "/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/job_state.py", line 85, in _status_update
    self.last_event = JobEvent.from_string(new_state)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250316_02/home/rinkhals/apps/40-moonraker/moonraker/moonraker/common.py", line 111, in from_string
    raise ValueError(f"No enum member named {enum_name}")
ValueError: No enum member named heating
2025-03-17 08:20:06,718 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.files.metadata, Code: -32601, Message: Metadata not available for </useremain/app/gk/gcodes/Deckel_PLA_0.2_41m41s.gcode>
2025-03-17 08:20:36,810 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.files.metadata, Code: -32601, Message: Metadata not available for </useremain/app/gk/gcodes/Deckel_PLA_0.2_41m41s.gcode>
2025-03-17 08:20:51,972 [klippy_connection.py:_process_status_update()] - Klippy has shutdown
2025-03-17 08:20:52,097 [proc_stats.py:_handle_shutdown()] - 
```

basvd, 2025-03-17:
Looks like for some reason the JobEvent enum in Rinkhals is modified from the original. It's trying to access the PAUSED member, which does not exist (has been commented out): https://github.com/jbatonnet/Rinkhals/blob/master/files/4-apps/home/rinkhals/apps/40-moonraker/moonraker/moonraker/common.py#L119

DerGary, 2025-03-17:
Thanks for the hint, Disabling Moonraker and Moonraker proxy seems to be a workaround for now, I was able to print one hour without reboots.

dpeter99, 2025-03-21:
I think I just had a related stop in the print.

[image]

Here are the logs from the website

[klippy.log](https://github.com/user-attachments/files/19397144/klippy.log)
[moonraker.log](https://github.com/user-attachments/files/19397143/moonraker.log)

jbatonnet, 2025-04-09:
A lot of improvements have been integrated in Rinkhals to improve memory usage and CPU consumption.
Please try last versions and avoid using too many Moonraker clients (OctoApp, Mainsail, ...) at the same time.

closed 2025-04-09 by jbatonnet