jbatonnet/Rinkhals #71: Moonraker support for gcode >100 MB

jbatonnet, 2025-03-17:


zerograv90, 2025-05-11:
Just a question - what's the reason that Moonraker has a problem with such big files?
Connection/binding issue? 
I ran also into this problem with an 85mb file.
Complete printer freeze (at least the ui) after uploading via orca slicer.

zerograv90, 2025-05-11:
From logs looks like a Moonraker problem. `2025-05-11 11:06:31,254 [klippy_apis.py:start_print()] - Requesting Job Start, filename = DRILL SWORD-PNP_PLA_0.2_7h56m.gcode
2025-05-11 11:06:31,260 [kobra.py:mqtt_print_file()] - Trying to print DRILL SWORD-PNP_PLA_0.2_7h56m.gcode using MQTT...
2025-05-11 11:06:31,416 [kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_auto_leveling" => "True"
2025-05-11 11:06:31,545 [kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_vibration_compensation" => ""
2025-05-11 11:06:31,663 [kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_flow_calibration" => ""
2025-05-11 11:06:31,669 [web.py:log_exception()] - Uncaught exception POST /api/files/local (127.0.0.1)
HTTPServerRequest(protocol='http', host='192.168.0.147', method='POST', uri='/api/files/local', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/web.py", line 1790, in _execute
    result = await result
             ^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/application.py", line 1017, in post
    result = await self.file_manager.finalize_upload(form_args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 833, in finalize_upload
    result = await self._finish_gcode_upload(upload_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/file_manager/file_manager.py", line 918, in _finish_gcode_upload
    await kapis.start_print(upload_info['filename'], user=user)
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/klippy_apis.py", line 147, in start_print
    ret = await self.run_gcode(script)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/kobra.py", line 401, in run_gcode
    self.mqtt_print_file(filename)
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/kobra.py", line 221, in mqtt_print_file
    client.connect('127.0.0.1', 2883)
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 851, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.11/socket.py", line 836, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
2025-05-11 11:06:31,829 [application.py:log_request()] - 500 POST /api/files/local (127.0.0.1) [_TRUSTED_USER_] 462338.72ms
-------------------- Log Start | Sun May 11 11:09:24 2025 ------------------`

jbatonnet, 2025-05-11:
For some reason, Moonraker is loading the entire file in memory when you upload a file instead of streaming it to storage.
The Kobra printers have ~220MB of memory, with 80-120MB being used by the system processes. So you end up out of memory, freezes and all.

Moonraker is badly optimized, efforts would need to be put modifying Moonraker to improve that. Technically, it's entirely possible to stream the file.

zerograv90, 2025-05-12:
Okey understand. Thanks for the explanation 😊
So technically it's an out of memory exception.

wxiux, 2025-05-16:
is it possible to have a workaround? altho printing 100+mb gcode is not everyday task, the few times I encountered is so frustrating, especially when one didnt know the cause

jbatonnet, 2025-05-16:
Use a USB drive or use Anycubic Slicer for those times you need a bigger file

basvd, 2025-05-17:
felipe (evil and ugly) on Discord posted this patch which could potentially fix the problem: https://github.com/felipezbofh/Rinkhals/commit/9ba7738953e485f01e950ed5d10350d63ddd1dd4

zerograv90, 2025-05-17:
As far as I understood the problem is on Moonraker side. Moonraker tries to load the complete gcode file into memory which is limited. So it's up to the free ram for the maschine when this limit is reached.

So the question is - will set a system var of tmpdir changes the behavior in Moonraker? And what's currently in use for TMPDIR. 

I'm also with basvd - who ensures that this tmp filesystem is cleared.
Else the storage runs out of space 
I will have a look at the Moonraker source later.

basvd, 2025-05-17:
Moonraker handles the upload by writing to a temporary file and then moving that file to the actual destination. The temporary file is provided by [tempfile](https://docs.python.org/3/library/tempfile.html), which uses the TMPDIR, which is tmpfs and therefore consumes RAM.

I don't think it's a good idea to change TMPDIR for the whole system though, because other applications use it and may not clean up after themselves. But we can pass the custom TMPDIR to Moonraker only. Moonraker itself does use `tempfile` for other things as well, so there may be other consequences...

zerograv90, 2025-05-18:
Just an idea, when we can pass an own tmp dir to Moonraker, can't we create and cleanup a Moonraker temp dir before we start Moonraker? So there should be no problem with the storage space.
The question then is - will Moonraker work normally with a temp directory within storage.

basvd, 2025-05-18:
Some extra details in this moonraker issue: https://github.com/Arksine/moonraker/issues/763

jbatonnet, 2025-05-18:
A PR is being prepared here: https://github.com/jbatonnet/Rinkhals/pull/187
I'll give it a 1 / 2 days max else I'll just implement it in the app