jbatonnet/Rinkhals #80: Kobra S1 retract and extrude filament while paused at specific layer problem (no ACE)

George9957, 2025-03-18:
To review, Kobra S1 has two versions. The  S1 with manual filament loader and other S1C with ACE auto 4 color filament unit.  
When printing with with S1 you can pause at a given layer by using the Slicer Next to insert a pause in gcode at specific layer. This works ok while printing and at the specified layer printhead moves to poop chute and pauses. At this point user needs to Retract filament and replace it with another roll.  After that an Extrude command is needed for user to feed the new filament roll. 
Problem:  while the printer is in "pause state" the Retract and Extrude buttons on the LCD are greyed out and are are not working. Only thing that works is resume button. This problem needs to be fixed.
This function is commonly available on other printers from Bambu and Prusa. It is necessary if you print TPU in colors which cannot be printed with ACE. You need to be able to Retract filament and Extrude filament while paused with the manual loader.
I hope it can be fixed.
Thank you for all your efforts.  George

Koberum, 2025-03-18:
That's a cool feature! 
I think when is changing color in a normal multicolor print with ACE, first cuts the filament, then retracts it a bit, then will start loading the new filament and starts the poop process cleaning the nozzle, then the print is resumed.

George9957, 2025-03-18:
There is a manual filament cutter right below the extruder, but it is finger controlled and not controlled by printer. I use it before retracting filament by using Retract button on LCD .  The ACE changes filament a little differently. But ACE cannot do TPU , too soft filament.

Koberum, 2025-03-19:
Got it. Seems like the printer is cutting the filament automatically only during prints with ace. Without that, you have to do it manually.

George9957, 2025-03-19:
Yes, that's it.   Don't understand why Anycubic firmware does not activate the Retract and Extrude buttons on LCD when doing manual filament swap. Did they forget? Other printers, Bambu. Prusa, and Creative all have this feature. You need it to do for TPU and disable the ACE unit which can't handle TPU.  Hope someone hears this plea.

jbatonnet, 2025-03-19:
Hi, just to understand completely, is this behavior the same between Rinkhals and stock firmware?

George9957, 2025-03-19:
Thanks for clearing that up for me. What you suggest it a bit deeper than I wanted to delve into on this subject. If someone looks at it and provides a fix that would be great.  In any case, thanks, and I'll be monitoring for any further developments.

jbatonnet, 2025-04-16:
Closing this one as there's no progress for 3 weeks

closed 2025-04-16 by jbatonnet