jbatonnet/Rinkhals #81: Kobra 2 Pro - device_id file missing/missing in printer files?

zerograv90, 2025-03-19:
**Describe the bug**
Downgraded the Kobra 2 Pro to Firmware to 3.1.2.3 from 3.1.4.
Installed Rinkhals release [Rinkhals 20250318_01].
Installations ended with two beeps but the screen freeze. No automatic restart even after 10 minutes. 
Manual restart ends with a freeze on the initial loading screen. But SSH Access is available.
After ssh login this shows up:
cat: can't open '/useremain/dev/device_id': No such file or directory

Where can if find the device_id - why can't the device id be determinated?

If the usb stick is disconnected the normal firmware boots up without problems.

**To Reproduce**
See above 

**Logs / Additional information**
Installations Logs:
Wed Mar 19 18:38:09 CST 2025: 
Wed Mar 19 18:38:09 CST 2025: Starting Rinkhals installation...
Wed Mar 19 18:38:09 CST 2025: Backing up machine-specific files
Wed Mar 19 18:38:09 CST 2025: Installing Rinkhals version 20250318_01
Wed Mar 19 18:38:09 CST 2025: Copying Rinkhals files
Wed Mar 19 18:38:24 CST 2025: Copying Rinkhals startup files
Wed Mar 19 18:38:24 CST 2025: Installing Rinkhals loader as it is missing
Wed Mar 19 18:38:24 CST 2025: Removing update files
Wed Mar 19 18:38:25 CST 2025: Rinkhals installation complete, rebooting...

/userdata/app/gk/config/device.ini
[device]
env : prod
zone : global
machineId : 
machineMac : 

for the bellow tags there seems to be for some blocks an entry for
machineCode 
deviceUnionid
deviceKey

jbatonnet, 2025-03-19:
Hi, I see where this error about device_id comes from but it should not be an issue in your case.
Based on this log everything went okay. Can you generate a debug bundle as described on the main page please?
This will create a zip with all the interesting logs

jbatonnet, 2025-03-19:
3.1.4 is not supported because it has caused some issues.
For the frozen screen last release should fix it: https://github.com/jbatonnet/Rinkhals/releases/tag/20250319_01

closed 2025-03-19 by jbatonnet

zerograv90, 2025-03-20:
Thank you very much. Nice work :) This solved the problem with the loading screen. And thank you for this wonderfull project