jbatonnet/Rinkhals #82: Camera is not working when connected via octoapp on port :7126

lequim2010, 2025-03-20:
When I connect to octoapp on port 7126, which is the only one that worked for me, the camera is not working, I think that is because the port, but I don't know the way how to avoid it.

[image]

HarunKilic, 2025-03-20:
Would like to see gcode preview from OctoApp, here is some errors I’m receiving:

[image]

———————————————-

Would like to see the speed and flow from OctoApp, also set speed to Slow or Sport, like in Anycubic app (if possible ofc). 

[image]

———————————————-

Here is some errors from webcam view:

[image]

jbatonnet, 2025-03-21:
Coming in next release.
Speed and flow will be treated separately: https://github.com/jbatonnet/Rinkhals/issues/28

closed 2025-03-23 by jbatonnet