jbatonnet/Rinkhals #91: Device startup failed, and cannot bind 'tcp:5037'

TekoXVI, 2025-03-24:
After installing Rinkhals, I get error code 11407 device startup failed. I tried restarting the printer and removing the .disable-rinkhals file. When I run start-rinkhals, this is what happens:

> Stopping Anycubic apps...
Killing 383 (./K3SysUi)
Killing 456 (./gkcam)
Killing 382 (./gkapi)
Killing 351 (./gklib-a/tmp/unix_uds1printer.cfg)
> Fixing permissions...
> Preparing overlay...
> Starting SSH & ADB...
/!\ SSH is already running
/!\ ADB is already running
> Preparing mounts...
> Restarting Anycubic apps...
/!\ Timeout waiting for gklib to start
Rinkhals startup was stopped with the .disable-rinkhals file, trying to start ADB...
root@Rockchip:/useremain/rinkhals# install_listener('tcp:5037','*smartsocket*')
cannot bind 'tcp:5037'

I have also tried a complete printer reset and clean Rinkhals installation but the same thing happens.

jbatonnet, 2025-03-24:
Most of the time when you have error 11407  it's because of printer configuration.
This page lists every trick to get it working: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Printer-configuration

Also, the port 5037 error is completely unrelated, it's a command you typed in you console.

jbatonnet, 2025-03-24:
If that worked, yes it means your printer configuration was somehow modified / corrupted and the printer could not start properly.
I'm glad it worked and you found a way to restore your printer.
I'm going to close this issue. If you have any other question, feel free to answrer back or join our Discord!

closed 2025-03-24 by jbatonnet