jbatonnet/Rinkhals #92: Rinkhals turned off (or got deleted) upon cold reboot

stylon, 2025-03-24:
I just installed latest Rinkhals (from 2025-03-23) to my brand new Kobra S1 (FW 2.5.0.2) and did print a couple of models just fine directly from regular OrcaSlicer. During the setup I did reboot the printer a couple of times as the network caused me some unexpected headaches. Upon reboots I was able to see the Rinkhals logo for a few moments and after booting completed the menu showed the Rinkhals menu entry.

Then I turned the printer off and went for dinner and turned it on right after. It showed the Rinkhals logo again during boot, but got stuck there. After 30 minutes (did a telco in the meantime), I came back and noticed it was still stuck, so I power cycled the printer. It rebooted, but without the Rinkhals logo during boot and also missing the Rinkhals menu entry now. Firmware version is still 2.5.0.2. So, I guess, it didn't run an automatic upgrade to the already available 2.5.0.6.

Any idea what might have happened? Is my config (e.g. for Mainsail) that I already created gone? Should I just try to reinstall Rinkhals?

stylon, 2025-03-25:
I reinstalled Rinkhals via USB stick like I did initially and it seems as if Rinkhals preserved some settings. However, none of the previously working network services (ssh on port 22, Mainsail on 80 and Octoprint on 7125) can be reached anymore. Any idea?

Unfortunately, what makes network testing quite tedious is that the Wifi adapter of the Kobra S1 does not play nice with any of my Wifi options available (4 routers + 1 mobile phone + 2 laptops). I've never had issues like that with any Wifi device before and there have been like >100 devices in the past 7 years that connected all just fine to my hardware (mostly the routers).

stylon, 2025-03-26:
In the install.log I found this difference between the initial Rinkhals installation and the re-install:

> Installing Rinkhals loader as it is missing

vs.

> Rinkhals loader was detected, skipping installation

jbatonnet, 2025-05-11:
Closing this one, the progress is captured in #109

closed 2025-05-11 by jbatonnet