jbatonnet/Rinkhals #100: No printing status on printer after sending printing job via lan from orca slicer on kobra 2 pro

zerograv90, 2025-03-26:
**Describe the bug**
With Rinkhals 20250319_01 and Rinkhals 20250325_02 the printing status dialog does not open on the printer display after upload & print from orcasilcer via lan print on the kobra 2 pro printer.

At this point there is no option to set the z offset manually because Kobra 2 pro supports this only under the printing status display. Also it is not possible to stop the print via printer display - only switch off :)

Mainsail shows/works everthing as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start print from orcaslicer via lan print -> Slice plate -> Upload and Print
2. Look at printer display (kobra 2 pro)

**Logs / Additional information**
Created two log zip files
- before_print_debug-bundle_4294967295_20250326-132441.zip - before start print from orcaslicer
- while_print_debug-bundle_4294967295_20250326-133125.zip - after start the print from orcaslicer

[image]
[image]
[image]
[image]

[before_print_debug-bundle_4294967295_20250326-132441.zip](https://github.com/user-attachments/files/19468246/before_print_debug-bundle_4294967295_20250326-132441.zip)
[while_print_debug-bundle_4294967295_20250326-133125.zip](https://github.com/user-attachments/files/19468247/while_print_debug-bundle_4294967295_20250326-133125.zip)

jbatonnet, 2025-06-11:
Another lead would be to try using K3 binaries on the K2P. Some are starting fine, so if the hardware is close enough and th binaries have everything they need it might work

zerograv90, 2025-06-16:
The question is, is it worth the expense. 
Whats about this Printer/StartPrint? Is it a local call on the gkapi? Seems to be contain only the gcode filename.

The firmware from k3 could work at least for the mainboard - but i thing configuration needs some adjustment/replacement from the k2p and much time for testing this.

As you say, printing works from orca without the printer ui. Also send gcode file and start print localy on print works with ui of course. Or as an alternative cloud print from anycubic next slicer of course with ui.
Also the bed seems to be relativ stable. So autolevel and adjust z-offset is not necessary for every print. Also the auto level is damn slow on the k2p. Even with the 5x5 lagrange.
And last monitoring/setting are available with fluidd or mainsail.

So it's more a comfort/functionaly completnes thing.
As is opened the ticket i wasn't aware, that the k3 has an internal mqtt server implemented for the lan mode.

I'd like to do further research here, but sadly I don't have much time for it at the moment.
But maybe i can contribute something for the FAQ page so its a visible resstriction like - No Print Status UI on start print via Orca Slicer? An then describe that this only works with lan mode (and a hint that k2p doesn't have the lan mode)? As general entry for all printers?