jbatonnet/Rinkhals #105: Camera's work in a slicer Anycubic Next

MiraMaX166, 2025-03-30:
After the firmware, the camera in the slicer does not work. How to fix it?

dhedberg, 2025-03-30:
You need to disable mjpg-streamer in the Rinkhals configuration menu on your printer, it's one or the other.

Incidentally, at least on my kobra s1, you can still view the camera with mjpg-streamer disabled by instead streaming from `http://<your-printer-ip>:18088/flv` (works with mpv, haven't tested anything else).

jbatonnet, 2025-03-30:
Yes as it it right now, support for both Anycubic apps and Mainsail / ... is not possible right now.
But @dhedberg's answer gives me some ideas, maybe we could spawn a ffmpeg instance alongside mjpg-streamer with flv on 18088.
I'll keep this issue open with this in mind!

MiraMaX166, 2025-03-31:
> You need to disable mjpg-streamer in the Rinkhals configuration menu on your printer, you can only use one or the other.
> 
> Incidentally, at least on my kobra s1, you can still view the camera with mjpg-streamer disabled by instead streaming from `http://<your-printer-ip>:18088/flv` (works with mpv, haven't tested anything else).

Thanks to the buddy. It helped.

MiraMaX166, 2025-03-31:
> Yes as it it right now, support for both Anycubic apps and Mainsail / ... is not possible right now. But [@dhedberg](https://github.com/dhedberg)'s answer gives me some ideas, maybe we could spawn a [...]

Thanks for your work on the firmware. I wanted to clarify whether your firmware makes changes to the standard printer settings, is the factory configuration file (printer.cfg and etc.) modified?

MiraMaX166, 2025-04-01:
> It is changed, but very little is done there. I only append this file: https://github.com/jbatonnet/Rinkhals/blob/master/files/3-rinkhals/home/rinkhals/printer_data/config/printer.rinkhals.cfg
> 
> It allows Fluidd to display properly, but apart from that it uses the default printer configuration. And everything is done with overlays, so no modification is ever done to the original config

Understood. This is the right approach. Thanks for the answer.

bchoor, 2025-06-18:
Does this hack `http://<your-printer-ip>:18088/flv` still work? I'm on S1 running 2.5.3.1 and not seeing it show up in mpv, VLC, or other player.

doctorevil30564, 2025-06-25:
> Does this hack `http://<your-printer-ip>:18088/flv` still work? I'm on S1 running 2.5.3.1 and not seeing it show up in mpv, VLC, or other player.

tried last week to see if it would work, I couldn't get it to work either, so I am going to say they changed something in the stock firmware that is now not allowing it to work like that. It would have been the perfect way to be able to use the camera on the stock workbench if there was a need to switch back and forth between using mainsail or the workbench.