jbatonnet/Rinkhals #106: Multiple USB Cameras

st3g4n0, 2025-03-30:
If you have several USB cameras connected to the device, you would need to add them to the Fluidd or Mainsail panel, but there is no option because you don't know the URL, but the ID of the device will be displayed.

jbatonnet, 2025-03-30:
@st3g4n0 can you please generate a Debug Bundle as detailed on the home page? This will give me the information about the device and filesystem to add support

st3g4n0, 2025-03-30:
`root@Rockchip:/root# ls -al /dev/v4l/by-id/*
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-Sonix_Technology_Co.__Ltd._USB_2.0_Camera_SN0001-video-index1 -> ../../video13
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-Sonix_Technology_Co.__Ltd._USB_2.0_Camera_SN0001-video-index0 -> ../../video12
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_474E7D4F-video-index1 -> ../../video11
lrwxrwxrwx    1 root     root            13 /dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_474E7D4F-video-index0 -> ../../video10
root@Rockchip:/root# v4l2-ctl --list-devices
rkisp-statistics (platform: rkisp):
	/dev/video8
	/dev/video9

rkisp_mainpath (platform:rkisp-vir0):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7

USB 2.0 Camera: USB Camera (usb-xhci-hcd.0.auto-1.1):
	/dev/video12
	/dev/video13

C922 Pro Stream Webcam (usb-xhci-hcd.0.auto-1.2):
	/dev/video10
	/dev/video11

root@Rockchip:/root# ls -1 /dev/v4l/by-id/* | sort | head -n 1 | xargs -I {} v4l2-ctl -w -d {} --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x90
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
[... 873 lines cut ...]
crw-rw----    1 root     video      81,   2 video2
crw-rw----    1 root     video      81,   3 video3
crw-rw----    1 root     video      81,   4 video4
crw-rw----    1 root     video      81,   5 video5
crw-rw----    1 root     video      81,   6 video6
crw-rw----    1 root     video      81,   7 video7
crw-rw----    1 root     video      81,   8 video8
crw-rw----    1 root     video      81,   9 video9
crw-rw----    1 root     video      81,  10 v4l-subdev0
crw-------    1 root     root       10, 125 rga
crw-------    1 root     root      244,   0 mpp_service
crw-------    1 root     root      243,   0 vcodec
crw-------    1 root     root       10, 124 rknpu
drwxr-xr-x    2 root     root           260 mpi
drwxr-xr-x    4 root     root            80 v4l
crw-rw----    1 root     video      81,  11 video10
crw-rw----    1 root     video      81,  12 video11
crw-rw----    1 root     video     250,   1 media1
crw-rw----    1 root     dialout   166,   1 ttyACM1
drwxr-xr-x    4 root     root            80 serial
crw-rw----    1 root     video      81,  13 video12
crw-rw----    1 root     video      81,  14 video13
crw-rw----    1 root     video     250,   2 media2
drwxr-xr-x   23 root     root           355 ..
drwxr-xr-x   15 root     root          1780 .`

jbatonnet, 2025-03-30:
Ok, this commit implements multiple cameras: https://github.com/jbatonnet/Rinkhals/commit/a81b294d16dd435d6c7baa9d99f0a7ce112abac4
I created a build for you until next release: https://github.com/jbatonnet/Rinkhals/actions/runs/14157972147

I made it so /webcam and /webcam0 are the first camera, then /webcam1, /webcam2 and /webcam3 are the other connected ones if needed

closed 2025-04-07 by jbatonnet