jbatonnet/Rinkhals #110: Using the cancel button in Mainsail Dashboard causes the next print to not start

VigilantGardener, 2025-04-02:
After cancelling a print using the one-step cancel button in the Mainsail dashboard, the printer will stay in a "Cancelled" status when trying to send the next print.  Sending the next print a second time will cause the printer to print as normal.

I added the cancel button to the dashboard in Mainsail.  This option can be added by going to Interface settings, the double gear symbol on the top right, then to UI-Settings, and enabling the slider for "Display CANCEL_PRINT".

Slicer is Orca Slicer 2.3.0 and the option used to send the print to the printer is Print >> Upload and Print.

Printer is on Firmware 2.5.0.6 and running the 3-25 version of Rinkhals

I tried to gather a log file, but after several minutes with the USB stick in the printer nothing happened.

lequim2010, 2025-04-03:
Yes, they have to ve renamed

VigilantGardener, 2025-04-04:
This is a debug I ran after cancelling a print job mid print.

[mid-print canclled via mainsail - debug-bundle_1261071018_19700101-025754.zip](https://github.com/user-attachments/files/19608343/mid-print.canclled.via.mainsail.-.debug-bundle_1261071018_19700101-025754.zip)

VigilantGardener, 2025-04-04:
This debug was ran after sending another print failed to start the printer.  The dashboard in mainsail was still showing the cancelled status.  An extra step was taken, the name of the gcode file being uploaded to the printer was changed.

[Sending print did not start printer - debug-bundle_1261071018_19700101-030306.zip](https://github.com/user-attachments/files/19608397/Sending.print.did.not.start.printer.-.debug-bundle_1261071018_19700101-030306.zip)

VigilantGardener, 2025-04-04:
This last debug is after another print was sent to the printer.  This time the printer started as expected and the print was allowed to finish.  The debug was pulled after the printer completed.

[2nd attempt and printer completed as noral or as expected - debug-bundle_1261071018_19700101-031726.zip](https://github.com/user-attachments/files/19608552/2nd.attempt.and.printer.completed.as.noral.or.as.expected.-.debug-bundle_1261071018_19700101-031726.zip)

jbatonnet, 2025-06-13:
Okay, your debug bundle shows error 10101 is issued when printing after a print was cancelled.
CODE:10101 There already has a printing task

In Mainsail/Fluidd you need to clear the current print status before printing a new file.
I think this is a normal behavior in Mainsail/Fluidd.