jbatonnet/Rinkhals #115: Control Buttons missing in Mainsail on Rinkhals 20250405_01_test

MJFairbairn, 2025-04-06:
**Describe the bug**
Kobra 3 (No ACE) 2.3.8
The Homing buttons don't exist for me all ok in Fluidd 
20250319_01 is the last version that displays them and still does.

[image]

[debug-bundle_2349366929_20250406-131721.zip](https://github.com/user-attachments/files/19621565/debug-bundle_2349366929_20250406-131721.zip)

[install.zip](https://github.com/user-attachments/files/19621570/install.zip)

jbatonnet, 2025-04-06:
Thanks for the report.
It's a known issue, that's a bug in Mainsail. I'm waiting for their next release to update it!

closed 2025-04-06 by jbatonnet

MJFairbairn, 2025-04-06:
Cheers.  Otherwise this is a super upgrade to Rinkhals, lots of hard graft. 
Most grateful to you and all the folks for an excellent system.