jbatonnet/Rinkhals #117: Improve apps system

jbatonnet, 2025-04-06:
[ ] OTA system
[ ] Better memory management UI
[ ] Repo manifest
    [ ] Trusted repos
[ ] Chroot apps for security
    [ ] Flags to get full access
    [ ] Flags to get resource level access?
[ ] Apps checksum for integrity
[ ] SWU safety to install apps

basvd, 2025-04-06:
Some ideas:
- Apps could specify the minimum supported version of Rinkhals.
- Apps could specify dependencies on other apps and minimum supported version (for example if an app requires moonraker, it should say so).

jbatonnet, 2025-04-06:
They could include supported printers as well

QuinnDamerell, 2025-04-06:
This is a great list. If I can chime in about updates, it would be great if you could follow something similar to moonrakers update progress. It's basically that you look for a new release tag in a given GitHub repo, if found, update pull the repo to that tag and run a script to update deps or at least rerun pip on the requirements.txt. 

I realize that the printer might not be able to do all of thag, but either way the amazing docker containers you have setup I would think it could be done via GitHub actions. Or maybe each repo could add a GitHub action that builds a new firmware or makes a zip with the new release bits.

Yangyuanxin, 2025-04-10:
Of course, if the application you are currently testing is stable enough in terms of printing experience, I hope to collaborate with you in the future to iterate certain features on firmware to enhance the experience.