jbatonnet/Rinkhals #118: Printing from Orca dubbel print files.

MichiMess, 2025-04-06:
I don't know if it happens with other slicers but when I send a print job from Orca 2 files come in Klipper. One of a few seconds cancelled, the other the actual print job. Without M117 in gcode, the display wont count. See pictures. 

[image]

[image]

MichiMess, 2025-04-08:
Thanks, when I read that, it sounds pretty logical. Do you know how I can solve this?

michas79de, 2025-04-15:
Same for me.
(I'm not using the latest Rinkhals, but the previous version for my Kobra S1 combo with f/w 2.5.0.6. I'll update very soon.)

Every print job uploaded from the current Orca Slicer results in one g-code file, but two history entries.
The first one is always canceled after 6 s and 30 mm of filament. So for me it's a longer time but the same length. Maybe it has something to do with the filament change / initial extrusion with the ACE combo system?

Everything is fine, but keeping the history clean manually (keeping only real cancels) is a bit annoying. But it's nice to see, this issue is already listed and close to the to-do-list.

[image]

CLNilix, 2025-04-17:
Hi, i think also its because of the "print start routine" of the printer/ACE.
I don't use orca (or any other slicer) to upload the g-code files; I upload the files manually through the fluidd dashboard.
And i have also this canceled prints for every print job (30 mm, 10s).

jbatonnet, 2025-06-25:
Very interesting, I'll try to have a look. If this patch is working you must be pretty close to the root cause! Thank you

jbatonnet, 2025-06-29:
Fixed in https://github.com/jbatonnet/Rinkhals/releases/tag/20250629_01