jbatonnet/Rinkhals #120: S1 install - hear beep upon USB install but no progress bar

Tomadoggy, 2025-04-07:
First off thank you!  Followed instructions to install Rinkhals on my S1 currently running  V2.5.0.6.  USB directory and update.swu, hear the beep upon USB insert, but then nothing...no progress bar, no activity.  Any ideas?

TomasOlsson, 2025-04-07:
Had same issue. My issue was because the USB stick wasn't empty. And make sure that's the file is named update.swu

Tomadoggy, 2025-04-07:
Yes, directory on the USB is named aGVscF9zb3Nf and the Rihkhals file is named update.swu.  This directory and this one .swu file are the only items on the USB, however still no progress after the beep.  Thank you for the idea however!

Tomadoggy, 2025-04-07:
Re-downloaded the swu file...working now.

closed 2025-04-07 by Tomadoggy