jbatonnet/Rinkhals #127: Installation hangs

Joerg1309, 2025-04-09:
As you recommended, I have now first updated the printer with the original Anycubic 2.3.8.9. Then with the current Rinkhals version.
However, it now hangs during installation.

[image]

Joerg1309, 2025-04-09:
[Archiv.zip](https://github.com/user-attachments/files/19675469/Archiv.zip)

jbatonnet, 2025-04-10:
Yes so I found the issue and I'm trying to find ways to fix it.
If the installation lasts for more than 3 minutes it gets killed by a timeout in the system. When it happens, it usually does during the long copy leading to 80%.
I'm happy for you, but I have to fix that quickly!

jbatonnet, 2025-04-10:
Here is a temporary build that fixes the issue if you or anyone gets this problem again until next release: https://github.com/jbatonnet/Rinkhals/actions/runs/14389056162

Joerg1309, 2025-04-10:
Thank you for your good Support.

jbatonnet, 2025-04-24:
Available in last release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250424_01

closed 2025-04-24 by jbatonnet