jbatonnet/Rinkhals #129: Issues with saved bed meshes causing 11407

jbatonnet, 2025-04-10:


rs13tux, 2025-04-11:
I have a Kobra S1 with 2.5.1.6 Firmware on it. I think i have the same issue. What i did today:

- Installing latest Rinkhals
- After rebooting the printer everythink seems to work proper, but i can't see any hightmap. Then i tried to measure an new one via the web-interface. That seems to work, but i still couldn't see the hightmap.
- After a second restart, the heightmap was there, but not under the name I had given it, but as "default".
- Then i tryed to measure a second hightmap at a higher temperature. The measurement worked, but was again not saved under the name I used, so i tryed to rename it. In the console i could see, that the map was renamed, but not in the web-interface.
- So I restarted the printer a third time. --> Since then i get the 11407 Error.
- Reboot
- 11407 is away, but Rinkhals also.
- Then i flashed 2.4.8.3
- Reset to factory settings
- Upgrade to 2.5.0.6
- Upgrade to 2.5.1.6
- Try to install Rinkhals a second time
- Installation itself runs without any issues, but after the restart i immediately get the 11407 again.

I can't understand that at all, since I had previously completely reset the printer. Or does the reflashing and resetting not delete the "rinkhals"-folder?

jbatonnet, 2025-04-12:
Factory reset is not a real factory reset, but all information in order to recover are listed in this wiki page: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Printer-configuration#i-got-11407-or-my-printer-doesnt-boot-anymore

rs13tux, 2025-04-12:
Thank you very much,
i already found the solution at discord.
After deleting the wrong mutable Files everything worked.

Best regards,
Roland

closed 2025-04-12 by jbatonnet