jbatonnet/Rinkhals #131: SWU tool to completely uninstall Rinkhals

basvd, 2025-04-11:
Sometimes the best solution to firmware issues is to just start over completely. It could be helpful to have a simple tool that completely removes all Rinkhals installations and data in home/rinkhals.

jbatonnet, 2025-04-16:
I think I'll include that in the boot helper tool, some kind of omni tool to reset config, enable/disable / uninstall, ...

jbatonnet, 2025-04-27:
11407 can be caused by trying to save multiple bed meshes as well: https://www.notion.so/thedju/Printer-configuration-1e145e4b6fd880bfaee4c4d6e42421b2?pvs=4#1e145e4b6fd88085b580e54d0dc91d1a

MrManiacNF, 2025-04-27:
I didn't do that either.. I disabled rinkhals.. and was getting a error about have unofficial code on the printer.. so that is why I flashed earlier version firmware.. so the printer would then prompt me to update to latest. Unless the bed meshs made themselves .. I did not do them manually. I did manage to get it back booting fine. Also why I have you here... I want to thank you for your work on rinkhals.. freakin awesome stuff. I recommend it to anyone looking for more out of their printer.

jbatonnet, 2025-05-11:
Closing this one, the progress is captured in #109

closed 2025-05-11 by jbatonnet