jbatonnet/Rinkhals #132: Spoolman 's M555 does not work for Kobra 3 v2.3.8.9 Rinkhals 20250409_01

nolantern, 2025-04-11:
**Describe the bug**
Spoolman Macro M555 S=<number> does not change the spool anymore.
In past it worked. I currently can't remember the release / firmware.

**To Reproduce**
1. slice a file with an other spool id
2. print it
3. the M555 does not send the new spool id to spoolman

**Logs / Additional information**
Kobra 3 Combo 2.3.8.9 Rinkhals 20250409_01

[debug-bundle_4244714901_20250411-163640.zip](https://github.com/user-attachments/files/19710888/debug-bundle_4244714901_20250411-163640.zip)

Moonraker log:
`
2025-04-11 14:24:21,095 [klippy_connection.py:_execute_method()] - Error running remote method: spoolman_set_active_spool
Traceback (most recent call last):
  File "/useremain/rinkhals/20250409_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/klippy_connection.py", line 561, in _execute_method
    ret = self.remote_methods[method_name](**kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: SpoolManager.set_active_spool() got an unexpected keyword argument 'SPOOL_ID'
`

derveloper, 2025-04-11:
> Spoolman Macro M555 S= does not change the spool anymore.
> nolantern created an issue [(jbatonnet/Rinkhals#132)](https://github.com/jbatonnet/Rinkhals/issues/132)

I noticed this too yesterday and a couple of days ago. Restarting the printer fixes the issue somehow. My current print changed the filament according to what I set in the gcode.

nolantern, 2025-04-11:
I rebooted the printer and now I can change the filament from the mainsail console like M555 S=<number>.
Thank you @derveloper !

Here is an other debug bundle if someone wants to check it.

[debug-bundle_4244714901_20250411-171831.zip](https://github.com/user-attachments/files/19711658/debug-bundle_4244714901_20250411-171831.zip)

kwe1009, 2025-04-12:
Having the same issue.  Rebooting has no affect unfortunately.

jbatonnet, 2025-04-12:
Spoolman was fixed quite some time ago and not really touched thereafter.
Make sure your Moonraker has the [kobra] component included as it's the one providing the fix.

kwe1009, 2025-04-14:
> Spoolman was fixed quite some time ago and not really touched thereafter. Make sure your Moonraker has the [kobra] component included as it's the one providing the fix.

this is what I have in moonraker.conf:

`# This file is generated automatically on every Rinkhals startup
# To modify its content, please use the printer.custom.cfg instead

[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /tmp/unix_uds1
max_upload_size: 1024

[machine]
provider: none

[authorization]
cors_domains:
    *://my.mainsail.xyz
    *://app.fluidd.xyz
    *://*.local
    *://*.lan
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

[octoprint_compat]

[history]

[database]

[kobra]

[spoolman]
server: http://192.168.15.8:7912
sync_rate: 5

`

The spoolman piece is in the moonraker.custom.conf file.  

Spoolman works for me except for the automatic changing of spools.  

This is my printer.custom.conf:
`[gcode_macro SET_SPOOL]
gcode:
    {{ action_call_remote_method("spoolman_set_active_spool", params) }}
[gcode_macro M555]
gcode:
SET_SPOOL SPOOL_ID={{ params.S }}
`

kwe1009, 2025-04-18:
Got it working.  Just started from scratch and reinstalled everything.

closed 2025-05-12 by jbatonnet

nolantern, 2025-05-19:
To me the issue is not solved.
I uninstalled and reinstalled rinkhals and added the parts to moonraker.custom.conf and printer.custom.conf. I did not factory reset the printer.
Same as before: sporadically it works. Most times I have to reboot the printer serval times until the M555 works in the Mainsail console.
It would be nice if this would be fixed but I understand that this could be a specific problem with my printer and not a general rinkhals problem.

Here is an error from the moonraker.log:
```
[klippy_connection.py:_execute_method()] - Error running remote method: spoolman_set_active_spool
Traceback (most recent call last):
  File "/useremain/rinkhals/20250514_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/klippy_connection.py", line 561, in _execute_method
    ret = self.remote_methods[method_name](**kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: SpoolManager.set_active_spool() got an unexpected keyword argument 'SPOOL_ID'
```