jbatonnet/Rinkhals #133: Mobileraker won't connect to Kobra S1 Combo

dJOS1475, 2025-04-12:
@jbatonnet first up, fantastic project, I only got my printer yesterday and have "jailbroken" it today (AS is frustrating AF!) so I can use Orca Slicer.

The only issue I've run into is Mobileraker wont connect after jailbreaking - im running the default config with the only exception is turning off Fluidd (no point running an unused UI).

Heere is the error report from Mobileraker:

```
Exception:
 type 'Null' is not a subtype of type 'num' in type cast

#0      _$$ConfigStepperImplFromJson (package:common/data/dto/config/config_stepper.g.dart:17)
#1      new _$ConfigStepperImpl.fromJson (package:common/data/dto/config/config_stepper.freezed.dart:346)
#2      _$ConfigStepperFromJson (package:common/data/dto/config/config_stepper.freezed.dart:18)
#3      new ConfigStepper.fromJson (package:common/data/dto/config/config_stepper.dart:40)
#4      new ConfigFile.parse (package:common/data/dto/config/config_file.dart:77)
#5      PrinterBuilder._updateConfigFile (package:common/data/dto/machine/printer_builder.dart:250)
#6      PrinterBuilder.partialUpdateField (package:common/data/dto/machine/printer_builder.dart:220)
#7      PrinterService._parseObjectType (package:common/service/moonraker/printer_service.dart:560)
#8      PrinterService._temperatureStore.<anonymous closure> (package:common/service/moonraker/printer_service.dart:514)
#9      _LinkedHashMapMixin.forEach (dart:_compact_hash:726)
#10     PrinterService._parseQueriedObjects (package:common/service/moonraker/printer_service.dart:575)
#11     PrinterService._printerObjectsQuery (package:common/service/moonraker/printer_service.dart:599)
<asynchronous suspension>
#12     PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:183)
<asynchronous suspension>

Failed-Key: configfile 
Raw Json:
{"configfile":{"config":{"bed_mesh":{},"buttons":{},"controller_fan controller_fan":{},"cs1237":{},"display_status":{},"engine_mode":{},"exclude_object":{},"extruder":{},"fan":{},"fan_generic air_filter_fan":{},"fan_generic box_fan":{},"filament_hub":{},"filament_tracker":{},"flow_calibration":{},"force_move":{},"gcode_arcs":{},"gcode_macro CANCEL_PRINT":{},"gcode_macro WIPE_ENTER":{},"gcode_macro WIPE_EXIT":{},"gcode_macro WIPE_NOZZLE":{},"gcode_macro WIPE_NOZZLE1":{},"gcode_macro WIPE_STOP":{},"gcode_macro WIPE_STOP1":{},"heater_bed":{},"heater_fan extruder_fan":{},"input_shaper":{},"led_pin":{},"leviQ3":{},"lis2dw12":{},"mcu":{},"mcu nozzle_mcu":{},"ota_filament_hub":{},"pause_resume":{},"pid_calibrate":{},"printer":{},"printer_marco":{},"probe":{},"resonance_tester":{},"safe_z_home":{},"stepper_enable":{},"stepper_x":{},"stepper_y":{},"stepper_z":{},"timelapse":{},"tmc2209 extruder":{},"tmc2209 stepper_x":{},"tmc2209 stepper_y":{},"tmc2209 stepper_z":{},"verify_heater extruder":{},"verify_heater heater_bed":{},"virtual_sdcard":{},"gcode_macro pause":{},"gcode_macro resume":{},"gcode_macro cancel_print":{}},"save_config_pending":true,"save_config_pending_items":{"bed_mesh default":{"algo":"lagrange","max_x":"245","max_y":"245","mesh_x_pps":"2","mesh_y_pps":"2","min_x":"5","min_y":"5","points":"-0.757167, -0.551667, -0.360500, -0.295667, -0.182167\n-0.572500, -0.342000, -0.070167, -0.076667, -0.057000\n-0.491833, -0.204500, 0.028500, -0.011500, -0.110667\n-0.593333, -0.406667, -0.180333, -0.256000, -0.413167\n-0.770833, -0.615667, -0.462500, -0.467000, -0.733833","tension":"0.2","version":"1","x_count":"5","y_count":"5"},"leviQ3":{"auto_zoffset_on_off":"1","bed_temp_cfg":"55.000000"},"probe":{"z_offset":"-0.0975"}},"settings":{"bed_mesh":{"algorithm":"lagrange","bicubic_tension":0.2,"fade_end":0,"fade_start":1,"fade_target":0,"horizontal_move_z":3,"mesh_radius":0,"move_check_distance":5,"speed":300,"split_delta_z":0.0125},"controller_fan controller_fan":{"cycle_time":0.01,"fan_speed":1,"heater":["extruder"],"idle_speed":0,"kick_start_time":0.5,"max_power":1,"off_below":0,"pin":"PB5","shutdown_speed":0,"stepper":[]},"cs1237":{"dout_pin":"nozzle_mcu:PA6","level_pin":"nozzle_mcu:PA7","sclk_pin":"nozzle_mcu:PA5"},"extruder":{"control":"pid","dir_pin":"!nozzle_mcu:PA11","enable_pin":"!nozzle_mcu:PC13","filament_diameter":1.75,"gear_ratio":[],"heater_pin":"nozzle_mcu:PB8","inline_resistor":0,"instantaneous_corner_velocity":1,"max_extrude_cross_section":50,"max_extrude_only_accel":5321.6216073828855,"max_extrude_only_distance":100,"max_extrude_only_velocity":159.64864822148658,"max_power":1,"max_safe_temp":320,"max_temp":340,"min_extrude_temp":170,"min_safe_temp":170,"min_temp":-180,"nozzle_diameter":0.4,"nozzle_material":"brass","pid_kd":108.268,"pid_ki":0.698,"pid_kp":17.393,"pressure_advance":0.03,"pressure_advance_max":0.1,"pressure_advance_smooth_time":0.04,"pullup_resistor":4700,"pwm_cycle_time":0.1,"rotation_distance":6.5756,"sensor_pin":"nozzle_mcu:PA0","sensor_type":"NTC ZM3486 HNS112HC4570G0060Y3486","smooth_time":1,"step_pin":"nozzle_mcu:PA12"},"fan":{"cycle_time":0.005,"kick_start_time":0.5,"max_power":1,"off_below":0,"pin":"nozzle_mcu:PB6","shutdown_speed":0},"fan_generic air_filter_fan":{"cycle_time":0.00003333333,"kick_start_time":0.5,"max_power":1,"off_below":0,"pin":"PC7","shutdown_speed":0},"fan_generic box_fan":{"cycle_time":0.00003333333,"kick_start_time":0.5,"max_power":1,"off_below":0,"pin":"PC6","shutdown_speed":0},"filament_hub":{"action_timeout":240,"config_file":"/userdata/app/gk/config/ams_config.cfg","cruising_speed":200,"cutter_pin":"nozzle_mcu:PB1","cutter_position":1.5,"default_feed_speed":30,"default_unwind_speed":20,"delay_before_throw":3,"feed_max_length":2600,"flush_multiplier":1.5,"impact_position_x":260,"impact_position_y":25,"impact_speed":10,"pre_sweep_position_x":47,"pre_sweep_position_y":230,"recalibrate_period":7200,"serial":"","serial1":"","serial_prefix":"usb-ANYCUBIC_ACE","shake_filament_speed":100,"sweep_fan_speed":0,"sweep_position_x":47,"sweep_position_y":276,"sweep_reprint_extrude_len":0.9,"sweep_retract_len":0.4,"sweep_speed":10,"switch_pin":"nozzle_mcu:PB0","throw_impact_speed":200,"throw_position_x":47,"throw_position_y":276,"unwind_length_after_triggered":1100,"unwind_speed_old_ace":15},"filament_tracker":{"length_per_pulse":1.86532063806894,"safe_unwind_len":150,"signal_type":"adc","tracker_detect_pin":"PB0","tracker_encoder_pin":"PB1"},"flow_calibration":{"delta_press_max":60000,"fit_val_max":20000},"gcode_arcs":{"resolution":1},"gcode_macro cancel_print":{"description":"G-Code macro","gcode":"CANCEL_PRINT_BASE","rename_existing":"CANCEL_PRINT_BASE"},"gcode_macro wipe_enter":{"description":"G-Code macro","gcode":"G90\nG1 F36000\nG1 Y250\nG1 F8000\nG1 X81\nG1 Y273"},"gcode_macro wipe_exit":{"description":"G-Code macro","gcode":"G90\nG1 F8000\nG1 X83\nG1 Y250"},"gcode_macro wipe_nozzle":{"description":"G-Code macro","gcode":"WIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1\nWIPE_NOZZLE1"},"gcode_macro wipe_nozzle1":{"description":"G-Code macro","gcode":"G90\nG1 F8000\nG1 X96\nG1 X81"},"gcode_macro wipe_stop":{"description":"G-Code macro","gcode":"WIPE_STOP1\nWIPE_STOP1\nG1 X75"},"gcode_macro wipe_stop1":{"description":"G-Code macro","gcode":"G90\nG1 F8000\nG1 X72\nG1 X77"},"heater_bed":{"control":"pid","heater_pin":"PA0","inline_resistor":0,"max_power":1,"max_safe_temp":120,"max_temp":140,"min_extrude_temp":170,"min_safe_temp":25,"min_temp":-200,"pid_kd":634.454,"pid_ki":0.441,"pid_kp":33.536,"pullup_resistor":4700,"pwm_cycle_time":0.1,"sensor_pin":"PC0","sensor_type":"NTC 100K beta 3950","smooth_time":1},"heater_fan extruder_fan":{"cycle_time":0.01,"fan_speed":1,"heater":["extruder"],"heater_temp":50,"kick_start_time":0.5,"max_power":1,"off_below":0,"pin":"nozzle_mcu:PB9","shutdown_speed":0},"input_shaper":{"damping_ratio_x":0.1,"damping_ratio_y":0.1,"shaper_freq_x":60,"shaper_freq_y":60,"shaper_type":"mzv","shaper_type_x":"mzv","shaper_type_y":"mzv"},"leviq3":{"auto_zoffset_on_off":1,"bed_temp":55,"bed_temp_cfg":60,"expansion_factor":0.04,"extru_end_temp":140,"extru_in":35,"extru_in_speed":30,"extru_out":50,"extru_out_speed":3,"extru_temp":170,"homing_retry_count":10,"max_diff":5000,"noise_diff":1750,"rough_platform_offset":[160,120,10],"to_z_up_speed":15,"to_zero_speed":80,"z_up":20},"lis2dw12":{"cs_pin":"nozzle_mcu:PB12","spi_software_miso_pin":"nozzle_mcu:PB14","spi_software_mosi_pin":"nozzle_mcu:PB15","spi_software_sclk_pin":"nozzle_mcu:PB13"},"mcu":{"max_stepper_error":0.000025,"resta [... cut ...] -0.770833,-0.615667,-0.4625,-0.467,-0.733833]]},"bed_mesh \"default\"":{"points":[[-0.757167,-0.551667,-0.3605,-0.295667,-0.182167],[-0.5725,-0.342,-0.070167,-0.076667,-0.057],[-0.491833,-0.2045,0.0285,-0.0115,-0.110667],[-0.593333,-0.406667,-0.180333,-0.256,-0.413167],[-0.770833,-0.615667,-0.4625,-0.467,-0.733833]],"mesh_params":{"min_x":5.0,"max_x":245.0,"min_y":5.0,"max_y":245.0,"x_count":5,"y_count":5,"mesh_x_pps":2,"mesh_y_pps":2,"tension":0.2,"algo":"lagrange"}}}
```

[image]

Cheers Derek

jbatonnet, 2025-04-12:
I guess that's because of this property

[image]

jbatonnet, 2025-04-12:
It seems to be missing from the config json

jbatonnet, 2025-04-12:
Okay so I had a look at their code. The `microsteps` property isn't used anywhere, but deserialized and marked as required.
I know the Klipper documentation states that this parameter must be provided, but it would change nothing to their app if they modify this part.
I would prefer that they modify their code instead of rather larger modifications in Rinkhals if possible.
Try to open a ticket on their repo for this issue.

dJOS1475, 2025-04-12:
Cheers, I’ll reach out to the Mobileraker dev.

jbatonnet, 2025-04-16:
Should be fixed with https://github.com/Clon1998/mobileraker/commit/4fbada77c1cb05289efac733b3a921a230a6969e