jbatonnet/Rinkhals #136: Kobra S1 SCREWS_TILT_CALCULATE

Affectimus, 2025-04-15:
Good afternoon, Dear developers!
I modified the table on the Kobra S1 printer and now I have the opportunity to adjust it. I really miss the function of helping to adjust the flatness of the SCREWS_TILT_CALCULATE table, I could not find this command in the HELP. Can you tell me if there is such an opportunity to add this functionality to Rinkhals for Kobra S1? Or is this functionality already implemented and am I doing something wrong?

EikeKre, 2025-04-16:
Hey, I wanted to ask the same question days ago for the Kobra 3 as I updated the bed to be adjustable with screws. Currently I run the hole bed mesh callibration after every adjustment

jbatonnet, 2025-04-16:
Hi, Anycubic printers run a reimplementation of Klipper but not the full vanilla Klipper. If the command does not work it won't be available in Rinkhals.
If you want to contribute in the development of vanilla Klipper on the printers, please join our Discord server to help!

I guess you tried the command already, so I'm closing this one as not possible.

closed 2025-04-16 by jbatonnet