jbatonnet/Rinkhals #137: Automatic leveling with cold bed

Mikmikbt, 2025-04-16:
Hello, I installed the unlocked firmware for the S1 (latest available 2.5.1.6) and I find it really very convenient.
I would like to point out a possible bug.
When I start the automatic leveling, this starts as soon as the nozzle has reached 140 degrees. The bed is not heated and remains cold.
In this way, the levelling is not correct.
Furthermore, the mesh is not automatically saved.

jbatonnet, 2025-04-16:
Hi, how do you initiate the automatic leveling?

If you start it from the printer screen or from Anycubic apps, Rinkhals doesn't modify anything so it would be a Anycubic issue. 
If you use Mainsail or another client, we have some level of control over the leveling process and yeah I could tweak things.

jbatonnet, 2025-04-16:
Okay so automatic leveling before printing and automatic flow calibration are related to #124 
You created another issue #138 for time lapse I'll respond over there
AI detect is not available through vanilla Moonraker, but you can use OctoEverywhere for example or publish something in the app reposotory if someone can see how to develop an app for that!

Going back to the original issue with automatic levelling, I assume you meant using Moonraker.
The stock leveling goes to 60 bed and 170 nozzle, wipes the nozzle then go to 140 nozzle while the bed stays at 60.
I cannot test write now but I assume you mean doing the leveling using Moonraker doesn't follow the same setup?

Also for the auto-save, I'm not touching anything here, so to my knowledge this is the default Anycubic bahavior.

Mikmikbt, 2025-04-17:
Hi, I saw that the timelapse issue has been closed... Sorry there is no
solution.Regarding leveling, I refer to Mainsail via the "calibrate"
command in the "height map" tab I don't use Moonraker because there is no
way to configure it.It remains to discuss the operation of the ACE pro and
color management. Do you want me to open another post? Thanks again

jbatonnet, 2025-04-17:
Color management from Mainsail and other Moonraker clients is not part of the roadmap for Rinkhals yet.
Now I know some people are working on standardizing MMU in Mainsail, and someone is looking at that for the ACE, so we might see some progress soon!

Mikmikbt, 2025-04-17:
Where will I find this job for Ace pro? Directly in this GitHub?

jbatonnet, 2025-06-17:
Available in latest release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250617_01

closed 2025-06-17 by jbatonnet

Mikmikbt, 2025-06-17:
Which problems are solved of the ones I reported? Thanks